solvespace/solvespace #1111: Add "ghost mode" for solid groups (prevent linked STL mesh "force to triangle" whole model)

Symbian9, 2021-09-14:
### System information

- **SolveSpace version:** `3.0~`https://github.com/solvespace/solvespace/commit/6bc63e92b0d908aa751ae06063b2ec00fec4cc57
- **Operating system:** all supported

### Expected behavior

Add option ("ghost mode") to solid groups to exclude group solids/meshes from any boolean operations in model — if such mode would be turned "ON" on *STL mesh group* its mesh should be "ghost"/reference assembled group which should be fully ignored in boolean operations and should NOT requires to apply `force force NURBS surfaces to triangle mesh` to any other solid groups.

Also, in situations with assembled NURBS solid groups external `*.slvs` and `*.emn` files "ghost mode" also would be useful where linked SolveSpace part or IDF circuit board would be used only for reference purposes in whole assmebly, BUT any boolean operations on such solid groups are unwanted.

Also, as actually *STL mesh group* is hardcoded to be *forced to triangle* it would be better to **dim** `[v] force NURBS surfaces to triangle mesh` **string** (in activated state with checkbox) instead of fully hide this string in *STL mesh group*.

### Actual behavior

Actually after *linking STL mesh* whole model became permanently *forced to triangle* - *suppressing STL mesh group* does not un-force triangle mode for the rest of NURBS solid groups either.

### Additional information

phkahler, 2021-09-15:
An STL file IS a triangle mesh. There is no option to mix NURBS and triangles meshes and there won't be. If you link a mesh, you're going to get meshes after that group.

Symbian9, 2021-09-15:
> If you link a mesh, you're going to get meshes after that group.

Then at least **suppressing** *STL mesh group* should turns back to solid NURBS mode.

closed 2024-03-12 by phkahler