solvespace/solvespace #1112: Add line styles for linked IDF & STL

Symbian9, 2021-09-14:
### System information

- **SolveSpace version:** `3.0~`https://github.com/solvespace/solvespace/commit/6bc63e92b0d908aa751ae06063b2ec00fec4cc57
- **Operating system:** all supported

### Expected behavior

- Add new lines styles: `s{id}-#def-stl-mesh` and `s{id}-#def-emn-solid`.
- Add `opacity` option for STL and IDF groups.
 
### Actual behavior

Actually styles of linked `*.stl` and `.emn` are hardcoded and there are NO way to change color of its mesh/solid and its solid group opacity.

### Additional information

phkahler, 2021-09-15:
There are no lines on STL mesh to style. STL files will use the color specified in the file if there is one. I think there is already an open issue to override colors of linked files.

Symbian9, 2021-09-15:
> I think there is already an open issue to override colors of linked files.

In https://github.com/solvespace/solvespace/issues/1073 I originally requested for solid groups, but guess I will STL and IDF groups too iin its description.

closed 2021-09-15 by Symbian9