solvespace/solvespace #1114: Include full source distribution in v3.0 release assets

kwshi, 2021-09-15:
The [version 2.3 release assets](https://github.com/solvespace/solvespace/releases/tag/v2.3) included a [`solvespace-source.tar.gz`](https://github.com/solvespace/solvespace/releases/download/v2.3/solvespace-source.tar.gz) archive in addition to GitHub's automatically-generated [source code archive](https://github.com/solvespace/solvespace/archive/refs/tags/v2.3.tar.gz), which include sources for the vendored dependencies (zlib, libpng, freetype, etc.) _at the time of release_.  This ensures that building from the released source is reproducible/deterministic.

The version 3.0 release does not contain such a source archive; that means, even when building from the _versioned_ source code, we can only use the one [generated by GitHub](https://github.com/solvespace/solvespace/archive/refs/tags/v3.0.tar.gz), which does not contain the vendored dependencies, which we would have to fetch separately from their upstream repositories.  This limits build reproducibility since upstream dependencies may change/release new versions, potentially breaking attempts to build solvespace from _versioned_ source release.

It would be extremely helpful & appreciated, especially from the perspective of OS package distributors, if you could include a full source archive w/ vendored dependencies included, on the 3.0 release page!

Symbian9, 2021-09-15:
Duplicate of https://github.com/solvespace/solvespace/issues/217 (and probably https://github.com/solvespace/solvespace/issues/1028)

kwshi, 2021-09-16:
OK, #217 got closed without resolution but #1028 remains open, so I'm going to close this issue in favor of #1028 (and move discussion there).

closed 2021-09-16 by kwshi

ruevs, 2021-09-16:
And https://github.com/solvespace/solvespace/issues/874