solvespace/solvespace #1115: Allow perpendicular constraints for faces

phkahler, 2021-09-16:
### System information

- **SolveSpace version:** 3.x
- **Operating system:** ALL

### Expected behavior

Perpendicular constraints should be allowed between faces. This could be used in assemblies or when using Revolve or Helix groups to create 90 or 270 degree bends for example.

### Actual behavior

Perpendicular constraints only work for 2D entities, not faces.

### Additional information

Not sure what to do with the purple symbol shown for perpendicular constraints. Faces are internally represented by a point and normal which defines an infinite plane, but point is not required to be near the sketch entities (for Helix I don't think it gets translated along the axis?). A face normal is encoded in the entity "normal" but it is not a full quaternion, instead it is a unit vector stored in the x,y,z components of the quaternion. The constraint equation is then just  q1.x * q2.x + q1.y * q2.3 + q3.z * q3.z  along with the implicit =0.

Symbian9, 2021-09-16:
> Not sure what to do with the purple symbol shown for perpendicular constraints.

If possible, lets there would be single "perpendicular" symbol placed somewhere on *intersection edge* (in one or both planes) and this symbol should be movable along this *intersection edge*.

I imagine it as there should be single virtual (invisible) datum point constrained on both faces + there should comment `⊥` text constrained to this datum point.

Here is mockup SLVS (see each group names) - [perpendicular-faces.zip](https://github.com/solvespace/solvespace/files/7179276/perpendicular-faces.zip)

phkahler, 2022-06-02:
duplicate #1246

closed 2022-06-02 by phkahler