solvespace/solvespace #1117: Add FLTK GUI (alternative frontend)

Symbian9, 2021-09-19:
### System information

- **SolveSpace version:** `3.x+`
- **Operating system:** all supported

### Expected behavior

Add FLTK GUI for SolveSpace.

### Actual behavior

Actually SolveSpace GUI based on `GTK2/3+`, and there is some work on preparation for `GTK4+` (https://github.com/solvespace/solvespace/issues/853). But GTK itself has many issue (with its ABI, theming, etc) which depends on each Linux distro.

Also there are requests on `Qt5` GUI (https://github.com/solvespace/solvespace/issues/401) and `ImGui` GUI (https://github.com/solvespace/solvespace/issues/263).

### Additional information

- FLTK supported Platforms/OS: *Windows, macOS, Linux and other Unix, etc.*
  - https://github.com/fltk/fltk/
- Custom FLTK widget themes:
  - https://github.com/Rangi42/tilemap-studio (more screenshots: https://hax.iimarckus.org/topic/7691/)
  - https://github.com/roukaour/viz-brain-visualizer
- Custom FLTK color themes (`FLTK Dark Theme`):
  - https://groups.google.com/g/fltkgeneral/c/3A5VC_854ok/m/sDpJsmuLBAAJ
- Custom FLTK widget schemes:
  - https://non.tuxfamily.org/wiki/NTK
- Cairo widget for FLTK and Qt:
  - https://github.com/user1095108/cairowidget

phkahler, 2021-09-19:
No. Solvespace was originally written with FLTK and it was abandoned for native Windows, macOS, and GTK. The ImGUI discussion has been had any number of times. It offers the project absolutely NOTHING, same is true of FLTK.

The argument that using a different GUI toolkit will enable more developers is nonsense. Solvespace has several very complex pieces and doing anything requires much more than GUI programming.

People have advocated for Qt several times, and a proper Qt version might get accepted if there is a pull request. However, I think half those people think we should use Qt on all platforms, which is not going to happen. Or they may think it's going to make the toolbars blend in on their precious KDE - it wont. Nobody wants Qt bad enough to implement the *one file* needed to do it.

Next person to beg for any GUI toolkit is getting hit with the ban hammer.

Closing as won't fix. With extreme prejudice.

closed 2021-09-19 by phkahler

Symbian9, 2021-09-19:
Okay.

edoelas, 2022-01-22:
Not sure if this is the right place to talk about this but:
I would love to see a nice qt interface for solvespace and I have some experience working with Qt + C++ as well as UI design. I am not an expert by any means, but I would like to give it a try.  I am interested in discussing this option, is it possible to talk about this topic with a current developer of the project?

ruevs, 2022-01-22:
@edoelas here is the issue about QT https://github.com/solvespace/solvespace/issues/401

kk6mrp, 2022-01-22:
You might find this useful also (https://solvespace.com/forum.pl?action=viewthread&parent=3611):

> If you really want, look in the platform folder here:
> https://github.com/solvespace/solvespace/tree/master/src/platform

> There is a single file (plus whatever CMake config stuff) for each platform. The guigtk.cpp file for example is about 1500+ lines of code. You'll need to implement the equivalent for QT in a new file probably called guiqt.cpp

> Nothing outside that folder and the build configuration should need to be changed to build on linux. Oh, and if you want to do this, please do Linux first and dynamic link against the system installed QT. We don't build any GUI toolkit code with SolveSpace for any OS.

edoelas, 2022-01-23:
Thanks, I have looked in those threads, and based on how the project is defined I think there is no point in porting it to qt. What I was thinking was more about a remake of the UI, not to use a different framework for no reason. It seems that for what I have in mind it would be better to use solverSpace as a library and forget to merge it with the main project.
On the topic of the usability of the interface, I will give my opinion: I think that solveSpace is a great program but the interface is amusingly bad. I am sure that many users that could be using it do not do it because of the interface. If the point of solvespace is to be used a rework from scratch is needed.

phkahler, 2022-01-23:
> On the topic of the usability of the interface, I will give my opinion: I think that solveSpace is a great program but the interface is amusingly bad.

@edoelas  I hope what you mean is that the UI looks like some kind of toy paint program from the 1990's or something similar. I can accept that as a perfectly valid criticism. However, there are a lot of people who (aside from the appearance) love the simplicity and ease of use solvespace brings. Once you learn the shortcut keys, there is little need to even look at the toolbar, and the text window contains a lot of useful information and is organized reasonably IMHO. My biggest problem with the UI is that the text and icons cannot be resized, nor can the font be changed, and I might also prefer a single window.

Symbian9, 2022-01-23:
> However, there are a lot of people who (aside from the appearance) love the simplicity and ease of use SolveSpace brings.

Agreed. And I guess "distraction free" UI/UX is the key feature of SolveSpace.

> My biggest problem with the UI is that the text and icons cannot be resized, nor can the font be changed

Yep, this + there are no way to change position of toolbar & axis icon (only show or hide in hardcoded position) — myself I want to has an option to show toolbar from "Property Browser" inside main window: https://github.com/solvespace/solvespace/issues/857#issuecomment-749051510

Hopefully, sooner or later SolveSpace would get resizable vector icons on toolbars:

- https://github.com/solvespace/solvespace/issues/857#issuecomment-749051510

Otherwise, adding three sizes (`16x16`, `24x24` & `48x48`) for SolveSpace is trivial task & thus it may be easily solved;)

> Once you learn the shortcut keys, there is little need to even look at the toolbar

Beside I like shortcuts, there are few issues:

- Its 2022 and we are most of our devices are "touch input", rather than "hardware keyboard" — so SolveSpace should be usable for those who has touch screen too;
- Actually shortcuts are hardcoded, but it would better if there would be "Keyboard Shortcuts Settings" to allow users to customize shortcuts.

> and the text window contains a lot of useful information and is organized reasonably IMHO...

Yes, and changes in upcoming `3.1` release would has a little bit better organized "Property Browser", than it was in `3.0` and older versions of SolveSpace.

BTW, there are still lot of work on polishing "Property Browser" UI/UX, so it is even not its final look ;)

>  and I might also prefer a single window.

Yep, this is long awaited feature:

- https://github.com/solvespace/solvespace/issues/39

edoelas, 2022-01-23:
I think that there are more problems than just the looks of it. For example: is there a way to set absolute coordinates when drawing a rectangle? Can I press the rectangle shortcut, type a start point and draw a rectangle without using the mouse?

phkahler, 2022-01-23:
> I think that there are more problems than just the looks of it. For example: is there a way to set absolute coordinates when drawing a rectangle? Can I press the rectangle shortcut, type a start point and draw a rectangle without using the mouse?

That an interesting question. Short answer is no, but hear me out.

First, should a rectangle be specified using the coordinates of two points, or one point and a width and height? Solvespace allows either, but you need to specify the constraints on the diagram. There is also no such thing as a rectangle is solvespace. When you use the rectangle tool it creates 4 lines and a bunch of constraints. You can actually delete the H and V constraints and add some perpendicular/parallel or equal constraints to allow the rectangle to be rotated.

Solvespace exists in the space between a technical drawing tool like LibreCAD and a full 3D modeling tool. A lot of information should be on the drawing.

There's a video comparing solvespace and FreeCAD for making a compressor part involving a helix. At some point he just constrains the twist in solvespace on the drawing. But FreeCAD insists on having the user type the pitch into a properties window - after taking some time to calculate what that value should be. I found the difference very telling. A properties dialog kind of defines the "one true way" of defining things. OTOH sometimes you want to specify pitch, and at the time we couldn't do that, but it's now optional with the helix. I put it in the text window because I didn't know how to annotate it on the drawing.

You raise a good point. Perhaps we should add a constraint that sets (and shows) the X,Y position of a point.

I'm not trying to discourage you. I'd like to see more experiments in this space. I just dont like hiding information in the text window even if it is sometimes necessary. But I also cant say for certain what portion of solvespace users share this philosophy.

ruevs, 2022-01-23:
> You raise a good point. Perhaps we should add a constraint that sets (and shows) the X,Y position of a point.

`"Constrain | Lock Point Where Dragged"` shortcut key `]`:
>"Am I a joke to you?"

;-)

Symbian9, 2022-01-23:
> you want to specify pitch, and at the time we couldn't do that, but it's now optional with the helix. I put it in the text window because I didn't know how to annotate it on the drawing.

Need to check drawing standard, but is not it just as bolt marking via arrow with label `M10 × 1.5` (where `1.5` is 1.5mm pitch)?

edoelas, 2022-01-23:
There are so many things to talk about just on the topic of squares. What I am suggesting is not to introduce automatic constraint, is just being able of using the keyboard instead of the mouse to set the points of the square. Autocad uses some kind of fancy command line to archive this. It looks like some contextual Widget, but at the end of the day is just a CLI. 
[image]

Something like being able to write: `Square(0,0,100,100)` should be enough.

About how to define the square (corner and corner, corner, width and height etc.). Usually the CAD softwares allow you to decide how to define the rectangle. Here there are a couple of screenshots:
- Autocad Mechanical:
[image]

-Solidworks:
[image]

-Fusion 360:
[image]

Imagine you want to define a rectangle that has its center in a certain point. If you use the option of corner to corner, the one that SolveSpace uses, you will have to create a rectangle, find its center with two construction lines and then use two constraints to make the crossing point to be on top of your desired point. 
[image]

With the other cad software you can just define a square by its center and that's all. Way faster. You don't always need to have a fully constrained sketch, in fact, working with fully constrained sketches in AutoCAD is madness. 

Some software, as for example, SolidWorks, introduce automatic constraints by default. But I don't think it is hiding anything from the user. The same way that when you define a square it has vertical an horizontal constraints, if you define the same square based on its center it will have an additional element, a point that represents the center. And this point will act as a symmetry axis horizontally and vertically. This can speed up incredibly the design process.

ruevs, 2022-01-23:
> I think that solveSpace is a great program but the interface is amusingly bad. I am sure that many users that could be using it do not do it because of the interface. If the point of solvespace is to be used a rework from scratch is needed.

On the other hand I think SolveSpace is a great program *_largely_* because of its interface. As I stated somewhere else - this is not a smartphone chat app ;-)

> Once you learn the shortcut keys, there is little need to even look at the toolbar

Absolutely! In fact the first thing I do is hide the toolbar - I _***never***_ use it. Well I would use it if I was recording a video tutorial... maybe... :-)

Many (most) advanced programs that expect "professional"/dedicated users have powerfull keyboard interfaces. For example it is a true joy to watch an advanced Photoshop user work. Or a PC gamer play a complex game ;-)

The point is UI is a matter of taste, and I personally think a "clicky", " iconey", "toobary", "colorful" UI will make SolveSpace a different program. One that some users will want to use and some will not.

Symbian9, 2022-01-23:
> One that some users will want to use and some will not.

Why just not provide SovleSpace users both ways:

1. enable & customize toolbars;
2. fully disable and use only shortcuts on their wish?

Thus each user may individually choose use it using only mouse/touch screen OR hide all toolbars and use only keyboard ;)

ruevs, 2022-01-23:
> > One that some users will want to use and some will not.
> 
> Why just not give users control on SolveSpace UI & shortcuts:
> 
>     1. enable & customize toolbars;
> 
>     2. fully disable toolbars and use only shortcuts on their wish?
> 
> 
> Thus each user may individually choose use it using only mouse/touch screen OR hide all toolbars and use only keyboard (with optionally set shortcuts) ;)

Because the other reason I absolutely ***love*** SolveSpace is that it is a single 6MiB executable (3.5MiB on ARM Raspberry Pi - just compiled it yesterday). And what you are suggesting creates bloat/complexity/code in non-trivial amounts. And (depending on how it is implemented) may cause the program to loose it's "razor sharp focus" (for lack of a better expression).

ruevs, 2022-01-23:
By the way - has anyone looked at the size (and number of files) of the QT runtime (the part that will need to be distributed and installed)? It is grotesque compared to SolveSpace! And I traditionally like QT and have used it in the past :-)

phkahler, 2022-01-23:
@edoelas   Thanks for the pics, they are very informative. So having a nested toolbar might be a decent way to handle a lot more construction tool variants, and we don't have a way to do that now.  +1 for an existing toolkit.  I had already considered making a new variant of rectangle that leaves it free to rotate (I was using SS to arrange furniture and wanted to rotate things). However, those large panels full of endless options for specifying a rectangle are IMHO a disaster from a UI point of view. Again, that's just IMHO. But I wouldn't mind 2 or 3 options on the main toolbar.

BTW to create a center point for a rectangle (today) you can just draw one diagonal and constraint a point at its mid-point. I opened the highly controversial issue #1142 to make this even easier - draw the diagonal, select it and press m. Same would apply for all those options that constrain a mid-point of a side or the bottom - just select the side and press m. This seems much nicer than scanning a huge sub-menu of specific options for creating a rectangle. Oh and squares - just select 2 sides and press "q".

As for typing commands, I think it might be nice but 1) there really isn't an API defined so it's a big job and 2) it wouldn't require much GUI toolkit - unless you want like a multi-line terminal looking panel. I think creating that functionality should come before changing toolkits to accomodate it.

Again, these are just opinions and they are part biased by knowing we have limited resources maintaining this software.

@ruevs  I agree, and I prefer GTK over QT. Some day we will probably need a proper cross-platform toolkit again. I just didn't think we're at that point yet.

Symbian9, 2022-01-24:
> Because the other reason I absolutely love SolveSpace is that it is a single 6MiB executable
>
>  I agree, and I prefer GTK over QT. Some day we will probably need a proper cross-platform toolkit again. I just didn't think we're at that point yet.

@ruevs @phkahler, To be clear, all I said above, could be implemented with GTK and OpenGL, I guess.

So, there would be no "size increasing" issue at all.

edoelas, 2022-01-24:
On the topic of weight, what I can say is: in my company we are creating a product whose interface is way more complex than solvespace and we are using QT. We use a statically compiled version of QT and the weight of the final executable is 17 MB.

phkahler, 2022-01-24:
We are not going to adopt QT on all OSes. End of story. I'm not going to give reasons because I'm not here to have a debate.

Symbian9, 2022-01-24:
> We are not going to adopt QT on all OSes.

* in official SolveSpace repo ;)

Symbian9, 2022-01-25:
Meanwhile, *GTK3* almost ported to *Haiku* (support for X11/Xlib, Qt, OpenGL & Cairo already added last weeks):

- https://discuss.haiku-os.org/t/xlibe-an-xlib-x11-compatibility-layer-for-haiku/11692/103

So, SolveSpace, as GTK3/OpenGL/Cairo might be ported & packaged for *Haiku* sooner or later ;)

@phkahler, Lets SolveSpace be GTK-based, but anyway its UI/UX should be improved using GTK & OpenGL features.

FTR, LibreCAD ported to Haiku since 2013:

- *LibreCAD* `2.1.3` on HaikuPorts: https://depot.haiku-os.org/#!/pkg/librecad/haikuports/2/1/3/-/5/x86_64
  - https://twitter.com/haikuos_es/status/328775961145380864
  - https://twitter.com/threedeyes/status/858508640667529217