solvespace/solvespace #1119: Exporting to SVG does not retain groups

Efruit, 2021-09-24:
### System information

- **SolveSpace version:** 3.0 ~ 0eab7f78
- **Operating system:** Fedora 33

### Expected behavior

Exported SVGs should have SVG paths grouped by their Solvespace groups.

### Actual behavior

SVGs only contain paths and styling information.

### Additional information

Ideally, the [\<g> elements](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/g) should:
- Contain all elements relevant to that group
- Have an `id` attribute that corresponds to the `g000` identifier. Solvespace's rules regarding group names is more lenient than [SVG's rules for IDs](https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/id#usage_notes).
- ...accordingly, the group name should be included as a [\<title> element](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/title).

I tried to write a patch, but turns out my C++ is a little rustier than I had thought (no pun intended). The natural route looked like adding a pair of methods like `StartPath` and `FinishPath` to `VectorFileWriter`. After that, `ExportViewOrWireframeTo` would have to be updated to create isolated `SEdgeList`s and `SBezierList`s for each group.

Symbian9, 2021-09-24:
Yep, actual SVG is messy if try to edit it Inkscape.

But there is a big question on how SolveSpace should export *outlines*, side *solid edges*, *fills* of cube with cut. Its not as easy as defining group ID. In some cases when SolveSpace export 2D view of solid model (e.g. cube with cut), single closed path (cutted side face fill) may refer to various groups, where its edges loop (in 3D) may refer to few other.

- SLVS - [cube-with-cut_ambient-lightning-1.0.zip](https://github.com/solvespace/solvespace/files/7226159/cube-with-cut_ambient-lightning-1.0.zip)
  [image]

*Inkscape* (SVG editor) has few "Styles" (line, fill), "Layers", "Objects" and "Groups":

- https://wiki.inkscape.org/wiki/Style_Editor
- https://wiki.inkscape.org/wiki/DocumenLayers
- https://wiki.inkscape.org/wiki/Object_tree

Some details new "Layers and Objects" hierarchy coming for in *Inkscape* `1.2`:

- https://twitter.com/realdoctormo/status/1439569832853348354
- https://twitter.com/realdoctormo/status/1434614162026606595

And new "Pages" hierarchy actually WIP for Inkscape:

- https://twitter.com/realdoctormo/status/1441091497042972678

So, overall, I guess there could be two SVG exports:

1. Simple SVG (just for WEB);
2. Inkscape SVG (with all possible features for next editing with Inkscape). 

FTR, *"Export 2D View... > AutoCAD DXF (\*.dxf)"* results in layered DXF, where each layer is assigned to each style (not to group ID) & include stacked objects per such style.

hinell, 2021-09-25:
Hi. Web front end developer here. 

There is a [`<g>`] element for grouping elements in SVG, checkout docs. Cheers.

[`<g>`]: https://developer.mozilla.org/en-US/docs/Web/SVG/Element/g

phkahler, 2021-09-25:
> * Simple SVG (just for WEB);
> * Inkscape SVG (with all possible features for next editing with Inkscape).

Inkscape SVG format is standard web SVG format, unless they changed something recently.

This feature request is fine - making the SVG export more useful in different cases would be nice. It's probably going to need someone to step up and make it happen. @Efruit you might want to try again. Make a (wip) pull request so we can see what you're up to. I have never even looked at the SVG export code :-/

Efruit, 2021-09-25:
> Inkscape SVG format is standard web SVG format, unless they changed something recently.

Yes and no. It's SVG-based, but they extend it a bit to support things like layers. It's still compatible with all but the most picky SVG parsers, but there are minor differences.

> It's probably going to need someone to step up and make it happen. @Efruit you might want to try again. Make a (wip) pull request so we can see what you're up to. I have never even looked at the SVG export code :-/

Frankly, I don't understand the internals well enough to figure it out. The Q&A required to get me up to speed would probably take as much time and effort as someone who's already in-the-know doing it from scratch
...not to mention that I hardly know C++.

I'm happy to publish what I have and keep whacking at it, but caveat emptor.

Efruit, 2021-09-26:
@phkahler That took a bit longer than I had hoped, but it sort of works now.  I'm hesitant to put it up as a PR because I have no idea what I'm doing, and I don't want to pass this off as something that could actually be merged.

https://github.com/Efruit/solvespace/tree/svg-groups