solvespace/solvespace #1120: New sketch from point and normal does follow its parent transforms

Symbian9, 2021-09-25:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/6bc63e92b0d908aa751ae06063b2ec00fec4cc57
- **Operating system:** Debian 10 buster

### Expected behavior

If new sketch created from point & normal, this new sketch should follow transforms of parent point and normal.

### Actual behavior

Actually if create revolve and use normal from resulted revolve ends for new sketch - new sketch created in same plane as revolve end face, but any changes in parent normal does not applied to new sketch orientation & new sketch origin does not follow parent point changes:

https://user-images.githubusercontent.com/13188580/134754585-09178328-a152-420b-bf23-5ab306659694.mp4

### Additional information

phkahler, 2022-01-30:
@ruevs  regarding my fix for this, I think we can use Subtype::WORKPLANE_BY_POINT_ORTHO for this instead of having a new subtype WORKPLANE_BY_POINT_ORTHO. At least I think that subtype is new but I can't find the PR that allowed this new way of defining a workplane.

phkahler, 2022-01-31:
> But if I do - then yes - it seems to me we could remove `WORKPLANE_BY_POINT_NORMAL` and use `WORKPLANE_BY_POINT_ORTHO` instead by checking if `g.predef.entityB` is filled in and get the quaternion from there.

I'm not really excited about doing that either. How many places would need to check for entityB? Same number that would need to check for WORKPLANE_BY_POINT_NORMAL, but that seems less explicit about what's going on and why.

ruevs, 2022-02-01:
>> I am not sure I understand your question correctly.
>>
> > But if I do - then yes - it seems to me we could remove `WORKPLANE_BY_POINT_NORMAL` and use `WORKPLANE_BY_POINT_ORTHO` instead by checking if `g.predef.entityB` is filled in and get the quaternion from there.
> 
> I'm not really excited about doing that either. How many places would need to check for entityB? Same number that would need to check for WORKPLANE_BY_POINT_NORMAL, but that seems less explicit about what's going on and why.

Then I do not understand your question correctly. (btw I would also not remove `WORKPLANE_BY_POINT_NORMAL` )

ruevs, 2022-02-01:
I added a [commit](https://github.com/solvespace/solvespace/pull/1201/commits/2e553b310fecb0ec071d5719682112f34d5f3b82) fixing the bug I described [above](https://github.com/solvespace/solvespace/issues/1120#issuecomment-1025791804).

PR #1201 (2022-02-03): Fix normal issue, when a workplane is created from a point and normal…
… we were only storing the numeric normal so the plane (and subsequent extrusion) wouldn't follow if the sketch was on the face of a revolve. Now we store the handle to the defining normal it predef.entityB so it can update properly on regeneration.

Fixes #1120
closed 2022-02-03 by phkahler