solvespace/solvespace #1122: Update messages.pot template

andesfreedesign, 2021-09-26:
Add
**'Show E&xploded View'
'Dimensions in &Feet and Inches'
'Length / Arc Ra&tio'
'Length / Arc Diff&erence'**

Cheers!

phkahler, 2021-09-26:
Oh there's a build command that will update it. Just have to remember what it is...

andesfreedesign, 2021-09-26:
@phkahler 
Ok, thanks!
Do you plan to integrate the texts of the **Property Browser** to be able to translate them?

phkahler, 2021-09-26:
To udpate the .pot file: 
```
make translate_solvespace
```
I also had edit locales.txt and CMakeLists.txt to actually add the new translation. PR is on the way.

phkahler, 2021-09-26:
> Do you plan to integrate the texts of the **Property Browser** to be able to translate them?

Ouch. I'm not sure the text window is ready for that. The previous maintainer wanted to rework it first, and I agree that's probably a good idea, but I might not be the right one for that task. It would also be good to have the text and icons resizable. OTOH if I didn't speak english I'd be annoyed that solvespace is only half-translated  ;-)

closed 2021-09-26 by phkahler

andesfreedesign, 2021-09-26:
@phkahler 
The es_AR.po file was not merged properly
Can you remove it and make a PR with the translations of the updated messages.pot?