solvespace/solvespace #1124: NURBS fails between Extrude and Revolve solids with same Circle profile

Symbian9, 2021-09-27:
### System information

- **SolveSpace version:** `3.0~`https://github.com/solvespace/solvespace/commit/6bc63e92b0d908aa751ae06063b2ec00fec4cc57
- **Operating system:** Debian 10.x buster

### Expected behavior

NURBS should not fails between Extrude and Revolve solids with same Circle profile.

### Actual behavior

Revolve NURBS operation fails between Extrude and Revolve solids with same Circle profile, BUT it happens if Revolve has specific angles according previous Extrude local Z-axis & Extrude profile local X/Y-axis

- SLVS: [tube-turns.zip](https://github.com/solvespace/solvespace/files/7235498/tube-turns.zip)

[image]

### Additional information

**NOTE:** Video would be added later today.

phkahler, 2021-09-27:
Is this the same as #171?  In that case the 2d coordinate systems for the circles are misaligned but the circles still lie exactly on each other.

If your "extrude path" is all in a plane (perpendicular to all Revolve axies) then this may be a different issue.

Symbian9, 2021-09-27:
> If your "extrude path" is all in a plane (perpendicular to all Revolve axies) then this may be a different issue.

This one.