solvespace/solvespace #1126: Active point does not follow mouse (cursor) postion in X or Y on canvas

Symbian9, 2021-09-27:
### System information

- **SolveSpace version:** `3.0~`https://github.com/solvespace/solvespace/commit/6bc63e92b0d908aa751ae06063b2ec00fec4cc57
- **Operating system:** Debian 10.x buster

### Expected behavior

If grab & move point, point should follows mouse (cursor) position (even if point is constrained on line, it should at least follow mouse X or Y position according canvas)

### Actual behavior

Moved point does not directly follows mouse X or Y position on canvas.

(I may be wrong, but it may be related to DPI someway: in SolveSpace `set to full scale` shows me `96 px/in`, but my monitor is highly likely is `~ 98 px/in`)

https://user-images.githubusercontent.com/13188580/134994406-f4531b95-60e8-4e66-89fd-476c06e8bb3b.mp4

One more issue here: when switching from orthogonal view to some isometric/rotated view active point may fully lost cursor position and follows mouse with some weird offset, and this does not reverted even after switching back to align view to workplane and adding next point (**additional video would be added later**).

[image]

### Additional information

phkahler, 2021-09-28:
> If grab & move point, point should follows mouse (cursor) position (even if point is constrained on line, it should at least follow mouse X or Y position according canvas)

It follows the closest position on the line to the mouse pointer, exactly as intended. You can also think of this as following the mouse position along (parallel to) the line.

When the sketch is rotated so the line is closer to parallel to the view direction (into or out of the screen) the same directional behavior is there, but the distance moved is reduced. It is in some sense "correct" if not what we always want or expect.

ruevs, 2021-09-28:
@Symbian9 How should it behave in the extreme case when you rotate the workplane that the line segment is in to be perpendicular to the screen (i.e. the line segment projected on the screen becomes a single point)?

What does Blender do in that case?