solvespace/solvespace #1138: Negative extrusion

shockergit, 2021-10-31:
Please add champering and filleting tool like the extrusion tool.

phkahler, 2021-10-31:
@shockergit  If by "Negative extrusion" you mean the ability to subtract an extruded shape from an existing model, just extrude your chamfer, go to the text window for that group and change "solid model as" from union to difference. In some cases doing an intersection might be better.

shockergit, 2021-10-31:
yes thank you soo much, thats all i wanted. and can you make that text window pinned on the right side  it would have been nice or it will overlap on the drawing window.

phkahler, 2021-10-31:
@shockergit  The text window is just a floating window. I think some people made experimental single-window attempt a few years ago. Not sure what came of it. I'm hoping to figure out how to get gnome to remember where the window was last time it was opened.

phkahler, 2021-10-31:
Yes, issue #39 which is very old, asks for a single window.

shockergit, 2021-10-31:
> 
> 
> Yes, issue #39 which is very old, asks for a single window.

Anyway thanks so much for this app its very easy and fun to use better than freecad in my case

ruevs, 2021-11-01:
https://github.com/solvespace/solvespace/issues/149

closed 2021-11-02 by ruevs