solvespace/solvespace #1142: Add midpoint when none selected

phkahler, 2021-11-05:
### Expected behavior

When the user selects a line and no point, pressing "m" should create a new point constrained to the midpoint of the line.

### Actual behavior

Nothing happens

### Additional information
This is not a bug, as "m" is meant to constrain an existing point to a line. Some users don't realize this and expect the midpoint function to *create* a midpoint, and that seems reasonable to do even if it's not quite consistent with usual application of constraints.

Symbian9, 2021-11-07:
> When the user selects a line and no point, pressing <kbd>M</kbd> should create a new point constrained to the midpoint of the line.

JFTR, What would happen if user would quickly set <kbd>M</kbd> few times in row?

Also, what about use next behavior instead:

1. Select *line* entity;
2. Press <kbd>P</kbd>;
3. Press <kbd>M</kbd>.

phkahler, 2021-11-07:
> What would happen if user would quickly or curiously press M few times in a row?

Well I hadn't thought of that. A naive implementation might create extra points at the midpoint. But only if the line remains selected?

Symbian9, 2021-11-07:
> But only if the line remains selected?

In theory, yes.

But I don't know, could be there a case where second <kbd>M</kbd> click in row would be faster then selected line would be unreleased after first  <kbd>M</kbd> click and creating dot on line in mid position? (on complex 3D models I few times spotted that selecting/unselecting may require a even 1-2 seconds)

JFTR, There is issue in "Property Browser" where ***if click*** on last (three-state) icon on toolbar (for switching show don't draw occluded/stipple occluded/draw occluded lines) ***too fast*** order of states is missed and ***it starts switch unorder***.

ruevs, 2021-11-08:
I don't particularly like either of the proposed ways to implement this.

Both break the "uniformity" of the UI and in my opinion it is much more important to have consistent behaviour without "corner cases" and "special cases" than to cater to a beginner.

Symbian9, 2021-11-08:
> I don't particularly like **either of the proposed ways** to implement this.

Should we vote on this feature to decide?

phkahler, 2023-01-02:
> I don't particularly like either of the proposed ways to implement this.
> 
> Both break the "uniformity" of the UI and in my opinion it is much more important to have consistent behaviour without "corner cases" and "special cases" than to cater to a beginner.

I agree. Closing this because it's just too weird at this time to add an entity from the constraint code.

closed 2023-01-02 by phkahler