solvespace/solvespace #1143: Properties of TrueType element show what the text was, not what it is

nabijaczleweli, 2021-11-08:
### System information

- **SolveSpace version:** df3ef2ab0e805b3500803607aa33e8152f11b048
- **Operating system:** Windows 10

### Expected behavior

The text in the property browser shows what it is.

### Actual behavior

The text in the property browser shows what it was.

### Additional information

An mpreg four illustrating the problem:

https://user-images.githubusercontent.com/6709544/140808519-be2abc13-5c4f-4626-8bdb-f91dcc129588.mp4

phkahler, 2022-01-17:
Can not reproduce this on Fedora Linux. Added the win32 tag because it seems OS specific.

ruevs, 2022-01-18:
@phkahler I can reproduce it on Windows but have not had time to debug it. Will try if #927 affects it bus as far as I remember there was something in that PR that I had doubts about.

ruevs, 2022-01-25:
Related (maybe the same)
https://github.com/solvespace/solvespace/issues/920
https://github.com/solvespace/solvespace/issues/863
https://github.com/solvespace/solvespace/pull/926

ruevs, 2022-06-12:
https://github.com/solvespace/solvespace/issues/920 is related.

closed 2022-06-14 by phkahler