solvespace/solvespace #1144: “STEP file doesn’t contain any solid bodies” according to Spanflug

barnabywalters, 2021-11-12:
### System information

- **SolveSpace version:** Version 3.0 (3.0~0e0b0252)
- **Operating system:** Mac OS 10.15.7

I’ll preface this issue with the following disclaimer: yes, I know this is very specific and probably more to do with a particular manufacturer than Solvespace itself. I don’t expect to receive much help with this, as I know the devs have much more important things to work on. However, I wanted to document this behaviour somewhere, so that at best it might get solved, and at worst will be a warning to others in my position. This seemed like the best place for it.

----

When using Solvespace to model a series of small, simple parts to get CNC-manufactured, I hit up against the following error when trying to get an automated quote from Spanflug, a German CNC machining company:

> “Die STEP-Datei enthält keinen Festkörper.“ i.e. “The STEP-File doesn’t contain any solid bodies.”

I had exported a “3D Surfaces” STEP, and my parts displayed just fine in their online 3D viewer, but for whatever reason their backend was unable to analyse the parts and give me a price. Their support people just told me to export the parts again, or use a different piece of software. I ended up using Fusion 360 to model my parts, which worked fine, but I wanted to figure out why their software didn’t like STEP files from Solvespace (which I much prefer using).

I modeled a simple test case, a 10mm x 10mm x 10mm cube, in both Solvespace and Fusion 360. As with my more complicated parts, the Fusion 360 STEP worked fine, but no solids were found in the Solvespace STEP.

I dug through the raw STEP files to try to compare them, and they’re quite different, even for this extremely simple object. Unfortunately, I don’t have enough experience with the STEP to have any idea what might be causing this problem.

If anyone has an idea what could be happening, or starting points for what things to edit in the STEP file to try to get them working with this manufacturer, I’d greatly appreciate it.

[Solvespace vs Fusion 10mm cube STEP files.zip](https://github.com/solvespace/solvespace/files/7530910/Solvespace.vs.Fusion.10mm.cube.STEP.files.zip)

ruevs, 2021-11-13:
@barnabywalters you are absolutely right. Here is the issue tracking this problem https://github.com/solvespace/solvespace/issues/206 .

ruevs, 2021-11-13:
You should at least look through the issue pinned at the top https://github.com/solvespace/solvespace/issues/1024 before creating a new one.

closed 2021-11-13 by ruevs

ruevs, 2021-11-13:
Hey this Spanflug place looks pretty nice. Good to know.

barnabywalters, 2021-11-13:
@ruevs oh thanks, I didn’t realise that was the same issue. Good to know it’s already being worked on!

I don’t have any experience with Spanflug’s parts myself yet, but a friend recommended them to me after a lot of good experiences, and they seem to offer better features than many similar services for quite reasonable prices.