solvespace/solvespace #1146: CVE's in `libdxfrw` (CVE-2021-21898, CVE-2021-21899, CVE-2021-21900)

Symbian9, 2021-11-21:
### System information

- **SolveSpace version:** `master`
- **Operating system:** All supported

### Expected behavior

Critical patches to LibreCAD's `libdxfrw` should be ported into SolveSpace's `libdxfrw`

- https://github.com/solvespace/libdxfrw

### Actual behavior

TL;DR: Multiple code execution vulnerabilities in LibreCAD's `libdxfrw`:

- https://blog.talosintelligence.com/2021/11/libre-cad-vuln-spotlight-.html

There are already patches for mentioned CVE's commited in LibreCAD's `libdxrw` repo:

- `CVE-2021-21898` patch: https://github.com/LibreCAD/libdxfrw/commit/ba3fa95648bef948e008dfbdd31a4d21badd71f0
- `CVE-2021-21899` patch: https://github.com/LibreCAD/libdxfrw/commit/6417118874333309aa10c4e59f954c3905a6e8b5
- `CVE-2021-21900` patch: https://github.com/LibreCAD/libdxfrw/commit/fcd977cc7f8f6cc7f012e5b72d33cf7d77b3fa69

### Additional information

phkahler, 2021-12-06:
Does anyone know why solvespace keeps its own version of libdxfrw? It seems like any PRs relating to that should just attempt to switch back to an upstream version if possible.

Symbian9, 2021-12-06:
> Does anyone know why solvespace keeps its own version of libdxfrw?

Guess, at least @jwesthues may/should know why it happen.

> It seems like any PRs relating to that should just attempt to switch back to an upstream version if possible.

IMSHO, SolveSpace devs should in some way collaborate with LibreCAD devs and unify `libdxrw` code base into single `LibreCAD/libdfxrw` repo.

FTR, *LibreCAD* `2.2.0-rc3` released week ago with all those CVE's fixed:

- https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.0-rc3

jwesthues, 2021-12-07:
I'm afraid I don't know why. Perhaps @Evil-Spirit does?

kk6mrp, 2021-12-07:
Check out https://github.com/solvespace/solvespace/commit/20a041e0efd02b481b91c1df93d112150b09854f

Symbian9, 2021-12-07:
> It's not clear to me where "upstream" is. They appear to be on sourceforge, but then there's something here on github with no releases listed.

@lordofbikes, may You clarify which one is actual place of upstream `libdxfrw` used for LibreCAD during compile process?

- https://github.com/LibreCAD/LibreCAD/tree/master/libraries/libdxfrw
- https://github.com/LibreCAD/libdxfrw
- https://github.com/codelibs/libdxfrw

phkahler, 2021-12-09:
Another option is to push ahead with #326 and use libreDWG instead. It's come a long way since that issue was opened, but the solvespace integration branch of @rurban had its last commit in January and I'm not sure where it stands.

Symbian9, 2021-12-09:
> and use libreDWG instead

And here is a thread on LibreCAD (GPLv2) + LibreDWG (GPLv3)

- https://github.com/LibreCAD/LibreCAD/issues/1055#issuecomment-451304367

phkahler, 2021-12-10:
SolveSpace is GPLv3+ and so is LibreDWG. That's awesome because we can statically link it into a single executable and continue not needing an installer for Windows ;-)

I do find LibreDWG's choice of GPL rather than LGPL a bit puzzling, but they are the FSF and I understand that hard line. At least they can change license later since they require a CLA.

I also wasn't aware of LibreCAD 3 untill two days ago, and then wasn't aware that it's a large rewrite - enough to change to GPLv3. If that's true, we have a nice constraint solver they could integrate ;-)  Maybe that's why they've been ignoring all those pull requests.

Symbian9, 2021-12-10:
>  At least they can change license later since they require a CLA.

Not good.

Does CLA is required to sign if fork/bundle LibreDWG under terms of GPLv3+ into other GPLv3+ software?

phkahler, 2021-12-10:
> Does CLA is required to sign if fork/bundle LibreDWG under terms of GPLv3+ into other GPLv3+ software?

No. You can mix and match any code under the GPLv3 license. The terms of use are totally "compatible" since the two projects are under the same license. Some licenses are only compatible in one direction, since the terms of use of one are a subset of the terms of the other. Most license pairings are incompatible (thanks OSI /sarcasm/) so it is important for FLOSS projects to keep to a small set of licenses that cover the important cases but no more.

The point of a CLA is to give a central person/organization the ability to change license in the future if the need arises. For example if FSF wants to change LibreDWG to LGPL or to GPLv2+ they could do it since they require a CLA for all contributions. That license change would allow FreeCAD to use LibreDWG directly as part of the software rather than keeping it as a separate external program.

rurban, 2021-12-10:
Unfortunately I don't have that much time anymore to work on that integration. 
But it's mostly a C++ solvespace job anyway, not so a LibreDWG job.

RE: License. The FSF made it very clear that GPLv3+ will stay. No LICENSE changes. But they own the code, so they could change it. I wouldn't be happy though. The adversary parties are huge and evil.

rpavlik, 2022-01-02:
Interesting. I actually went thru and fixed some issues in dxfrw, realized they had a separate git repo (the codelibs one), and submitted our stuff plus a few more cleanups/modernizations: https://github.com/codelibs/libdxfrw/pull/19 and made some additional cleanups here: https://github.com/codelibs/libdxfrw/pull/20

Symbian9, 2022-01-26:
Additionally to already fixed `CVE-2021-21898`, `CVE-2021-21899`, 
`CVE-2021-21900`, now there are few more CVEs:

- https://cve.report/CVE-2021-45341
  > A buffer overflow vulnerability in CDataMoji of the jwwlib component of LibreCAD 2.2.0-rc3 and older allows an attacker to achieve Remote Code Execution using a crafted JWW document.
- https://cve.report/CVE-2021-45342
  > A buffer overflow vulnerability in CDataList of the jwwlib component of LibreCAD 2.2.0-rc3 and older allows an attacker to achieve Remote Code Execution using a crafted JWW document.
- https://cve.report/CVE-2021-45343
  > In LibreCAD 2.2.0, a NULL pointer dereference in the HATCH handling of libdxfrw allows an attacker to crash the application using a crafted DXF document.

lordofbikes, 2022-01-27:
These 3 vulnerabilities are not related with `libdxfrw`, only with LibreCAD!
So, don't panic solvespace :wink:
Also all three are already fixed.

rurban, 2022-01-27:
Also libredwg has plenty of such buffer overflows, as it's continuously fuzzed by google osfuzz service. 
It also has a much larger attack surface because it supports 220 more object classes.