solvespace/solvespace #1147: SolveSpace Official Accounts on social networks

Symbian9, 2021-12-07:
### System information

- **SolveSpace version:* all
- **Operating system:** all

### Expected behavior

Dear, @jwesthues @phkahler @ruevs @ppd.

For a long time since SolveSpace `2.0` released as open-source projects it was less highlighted for FOSS/FLOSS community, and even after `3.0` release it is still yet obscure project.

*I want ask you all to **grant me a permission to represent SolveSpace Team on various social platforms** (for development news highlight & users support purposes).*

I'm SolveSpace *user* since 2013 & active *tester / contributor* since 2015. Also *posted news* on SolveSpace releases on various Linux-related blogs & platforms (such as HN, Reddit, Twitter, YouTube, etc.) and is an editor of SolveSpace-related articles on Wikipedia. Since November 2020 I'm a *moderator* of `r/SolveSpace` sub on Reddit.

Actually I'm *trying to showcase* latest SolveSpace features & trying to provide support for SolveSpace users via Twitter, Reddit, Mastodon, YouTube and on other platforms.

My showcase repo (+ links to my accounts on platorms):

- https://github.com/Symbian9/SolveSpace-Daily-Engineering

Please, allow me be a member of the official SolveSpace Team as its "voice on social networks")

### Actual behavior

Actually there are no any "official" SolveSpace accounts of on the most popular social platforms (Twitter, Reddit, YouTube, Mastodon, PeerTube, etc.), and as a result SolveSpace is yet obscure project.

On Twitter `@solvespace` was registered by someone else not related to SolveSpace at all.

On Reddit there is already `r/SolveSpace` sub (actually not yet official) created by @dotancohen on June 7, 2020 (and I'm one of its since November 2020), see:

- https://github.com/solvespace/solvespace/pull/1141

### Additional information

ruevs, 2021-12-09:
Hi @Symbian9 ,

  In my opinion your (and anyone's) efforts to popularize SolveSpace are a great thing!

  As to your request
> to **grant me a permission to represent SolveSpace Team on various social platforms**

What do you mean specifically? SolveSpace is open source and (at least I) certainly can not "grant" you or anyone a permission to popularize it on social networks. It's not forbidden in the first place. At least that's what I think.

  In fact I'm only answering because you addressed me directly in your request.

ruevs, 2021-12-09:
As for:

> Please, allow me be a member of the official SolveSpace Team as its "voice on social networks"

If you mean the team here on the GitHub repo - that is not my decision, but I think @phkahler  added the rest of us based on a very simple, purely technical criteria - just look here https://github.com/solvespace/solvespace/graphs/contributors ;-)

Symbian9, 2021-12-09:
> If you mean the team here on the GitHub repo

No, I mean just recognize me as member of SolveSpace project in general meaning, not on GitHub :)

> What do you mean specifically? SolveSpace is open source and (at least I) certainly can not "grant" you or anyone a permission to popularize it on social networks

I mean I wold be happy to create official SolveSpace app acconts on Twitter/Mastodon, also it would be cool if  SolveSpace Team allow to claim `r/SolveSpace` as official subreddit of SolveSpace project on Reddit ;)

phkahler, 2021-12-09:
I want to close this without comment, but here I am. I don't think SolveSpace needs an official social media representative at this time. It does need more developers and continued promotion on other sites, and we should all continue to bring it up in appropriate ways when opportunity comes along.

I find the wording odd too "permission to represent SolveSpace Team on various social platforms". No, I don't want you to represent me or anyone else.

Symbian9, 2021-12-10:
> It does need more developers and continued promotion on other sites

As SolveSpace is almost is uknown, I can't see the other way to join active FOSS/FLOSS software devs instead of promote it

As for now there are two major users of SolveSpace who *actively promote use of SolveSpace app and create showcases examples & videos*:

- me, on Twitter: https://twitter.com/app4soft
- and Eric Buijs on Mastodon (Fosstodon): https://fosstodon.org/@ericbuijs

And there are plenty (I know not more than 20) SolveSpace users who use AND/OR showcase SolveSpace publicly.

>  No, I don't want you to represent me or anyone else.

To be clear, I under *"permission to represent SolveSpace Team on various social platforms"*  I just mean that I want represent SolveSpace app as its contributor & team member and then `r/SolveSpace` could be claimed as official & I want to create and manage official SolveSpace account on Twitter as Team  account (separately from my personal accounts & without my POVs) and manage it same way as FreeCAD's https://twitter.com/FreeCADNews

- https://blog.twitter.com/en_us/a/2015/introducing-tweetdeck-teams
- https://help.twitter.com/en/using-twitter/tweetdeck-teams

> I don't want you to represent me or anyone else.

@phkahler, but why? FTR, I do NOT want claim my personal accounts (`app4soft`) as official SolveSpace - just want manage some official Twitter/Mastodon & Reddit, as it is the most popular platforms used by Blender, FreeCAD, QCAD & other FLOSS/FOSS apps devs & teams for promote its free & opensource software.

ruevs, 2021-12-10:
> I want represent SolveSpace app as its contributor & team member

@Symbian9  But you re not a team member. So it would be... misleading... to represent yourself as one. So I do not "give you" permission to do so.

Your attempts to organize things are... unproductive... for the scale of the SolveSpace project. Just continue to popularize it in any way you wish. Saying "I'm a team member" won't make it more effective.

You are an active, very skilled user, and I for one appreciate your bug reports and some of the [example files](https://github.com/solvespace/solvespace/issues/1091#issuecomment-945190823) you post from time to time. By all means continue to do it.

Symbian9, 2021-12-10:
> So it would be... misleading... to represent yourself as one
>
> Saying "I'm a team member" won't make it more effective.

No. Again, I just clarified that I just want to create SolveSpace account on Twitter as official & set /r/SolveSpace on Reddit as "SolveSpace User Support Team")

Anyway, I'm officially `Contributor` to SolveSpace development already, so I'm part of SolveSpace Contributors Team.

> If you mean the team here on the GitHub repo - that is not my decision, but I think @phkahler added the rest of us based on a very simple, purely technical criteria - just look here https://github.com/solvespace/solvespace/graphs/contributors ;-)

To be more clear, https://github.com/solvespace/solvespace/graphs/contributors does NOT represent issues/bug reports submitted by user and feature requests/ideas that user proposed, but implemented by other.

I'm contributor, except localizing, to SolveSpace QA/testing and UI/UX improvements ideas, but it is not shown on the "Contributors Graphs" — of course, coding is a main part development, BUT testing/QA issues submitting & idea proposing (which may be implemented by coders) is also important part of contributing to software development.

- My issues: https://github.com/solvespace/solvespace/issues?q=is%3Aissue+author%3ASymbian9
  - Comments to other issues: https://github.com/solvespace/solvespace/issues?q=is%3Aissue+-author%3ASymbian9+Symbian9

ruevs, 2021-12-10:
I said you are not a team member - @phkahler (quite wisely in my opinion) seems to have chosen team members based on commits to the code (as in C++, programming). When writing about SolveSpace please do not make it sound as "official SolveSpace team" statements - there is no such thing - In my opinion we are too small to have this level of "formality":

```
The official SolveSpace team hereby announces that they have released SolveSpace version 10 - the latest and greatest ever. The days of commercial CAD are numbered. Surrender now!
```

:-D ;-)

Whenever I mention SolveSpace I start with:
 - "Do you know about this great, small fast free CAD tool?..."

or
 - "There is this amazing open source CAD program - SolveSpace - that I enjoy so much that from time to time I even write code to make it better..."

> I'm contributor, except localizing, to SolveSpace QA/testing and UI/UX improvements ideas,

Of course you are a contributor - no doubt about it - in all the ways you described. And I - personally (not as an "official team member :-D ) am perfectly fine with you saying something like "I'm a contributor to SolveSpace, have used it for many years, it's amazing, and here is this great model I just created, which shows how good SolveSpace is."

ruevs, 2021-12-10:
To make it even more clear - what you are suggesting makes sense for a "big enough project" where apart from the "mandatory":
 - programming
 - debugging
 - testing

There is "room" for formal "teams" for :
- QA
- marketing
- support
- process
- planning
- management
- PR
- accounting
- HR
- etc.

At this point SolveSpace is not big enough for anything from the second group.

I'll close this. Feel free to continue the discussion.

closed 2021-12-10 by ruevs

Symbian9, 2021-12-10:
> At this point SolveSpace is not big enough for anything from the second group.

I'm not agree with statement that SolveSpace is not big enough for anything from the second group.

- https://old.reddit.com/r/FreeCAD/comments/raz787/yorik_van_havre_on_twitter_i_really_need_to_have/

But I may agree that SolveSpace Forum (its look & login behaviour) is bad for normal support site.

> There is "room" for formal "teams" for :
>
> - QA
> - marketing
> - support
> - process
> - planning
> - management
> - PR
> - accounting
> - HR
> - etc.

Lets clarify each of them:

- **QA** - it is mostly all of users who reports SolveSpace issues and request improvements to existing features;
- ~~**marketing**~~ - NO, as SolveSpace is FLOSS app, there should not be "marketing" at all, BUT I guess there should be some ways to support SolveSpace Team core members via **donate** (Patreon, LiberaPay, etc.)
- **support** - that is a team of all SolveSpace users who help other SolveSpace users with any issues;  but as I sad, SolveSpace Forum is looking bad and should be redesigned;
- **process** - ?
- **planning** - that is probably the most major thing yet missed in SolveSpace - there are no long term planning for SolveSpace future;
- **management** - PM (Project Management) of which type of management?
- **PR** - that is exactly a team what I requested to be member of in actual issue thread; as for now it may include: *me* (@Symbian9), @ericbuijs and @mhalil as the most active SolveSpace users in PR;
- **accounting** - ?
- **HR** - joining  new developers to SolveSpace is really depend on popularity of software, so its depend on PR team work too;
- **etc.** - ?

ruevs, 2022-01-12:
Hey my "marketing efforts" paid off https://www.youtube.com/watch?v=CHfFfl_zFC8&t=1160s :-)

Symbian9, 2022-01-13:
> Hey my "marketing efforts" paid off

@ruevs, Oh, so it was You who triggered The Melt Zone!? Thank You so much!

Watched it few weeks ago & shared on Twitter: https://twitter.com/app4soft/status/1477675754427240456

ruevs, 2022-03-30:
@Symbian9 looking at https://mobile.twitter.com/SolveSpace it has turned into an endless feed of things completely unrelated to SolveSpace. Irrespective of whether I agree with them or not they are unrelated. And this post-factum answers your question why I personally did not agree for you to "represent" SolveSpace on social networks.

Keeping in mind what is going on - I hope you are fine and will be safe.

ppd, 2022-03-30:
Three opinions:
1. I think marketing/PR is very important, so it makes some sense to maintain a focused social media presence in general.
2. Nothing in life is truly apolitical, but a software project's social media account should mostly stay on topic.
3. Beyond simple statements of fact ("there is a new release"), only those who have strategic influence in the project itself can represent it in a meaningful way. I, as a contributor of very peripheral (and few) things, can't speak for the project. This does not mean the contribution is worthless, but that it's merely an enabler of the project's core functionality.

Therefore, personal ("fluff") social media accounts should be clearly distinguishable for the average reader from channels that are under the control of the core project leaders.

In any case, I hope you're safe @Symbian9.

phkahler, 2022-03-30:
@ruevs @ppd and anyone else interested. Just prior to all this, that twitter account made a post suggesting people donate to a patreon account of the same person, written as though they were a 3rd person. It looked like an offical solvespace account encouraging donations to a "contributor" who happened to be in Ukraine but was in fact the same person, who is not a significant contributor. Symbian has also doxxed whitequark in recent posts, and has also caused some animosity with one of the FreeCAD developers. I've been personally contacted by someone else to bring some of this to my attention. Our project is getting bad PR.

For all the reasons above, as well as past annoyance, he has been blocked from the Solvespace organization and its repositories on github for a couple week already. Don't bother tagging him and please don't bother talking about him. Unfortunately I have no way to claim what looks like an official project twitter account, and have not had time to setup a different one, much less post anything to it. @ruevs if you want to create/run one please contact JW, otherwise I'll try to get to it when the chaos in my own life eases up a bit.

Hopefully this will be my last comment on this issue.

ruevs, 2022-03-30:
@phkahler oh, what a sad mess, I wish I had not googled it :-( But there is an upside - now I am convinced you did well to ban him. To be honest I've been continuously surprised over the years that Whitequark did not ban him. As for the Twitter account called SolveSpace - that he controls... ehhh... that's unfortunate. Are there ways to "claim" official sounding accounts?

I don't have a Twitter account and almost never visit the site (took me more tan a month to notice this!) so I'd be the worst person to maintain a SolveSpace official one. Facebook maybe - but it's probably pointless.

JulianGmp, 2022-04-16:
Sorry for digging up this issue despite it being closed.
I just looked through twitter's support page and found this: https://help.twitter.com/en/rules-and-policies/twitter-impersonation-and-deceptive-identities-policy

I think if you wanted to, I think you could claim the `@SolveSpace` user for yourself. Though I'm not 100% certain how well twitter's support handles these cases.

ruevs, 2022-04-17:
Reading that policy at most the account may end up permanently disabled. Perhaps I missed it but there was nothing about "claiming" it.