solvespace/solvespace #1148: Solver fails to converge when applying parallel constraint to exactly perpendicular lines

henrebotha, 2021-12-10:
### System information

- **SolveSpace version:** 3.0~0e0b0252
- **Operating system:** Xubuntu 21.10

### Expected behavior

I imported `column.slvs` into `matrix-right.slvs` and attempted to set the normals of the centre point of the circle parallel to the same point on another circle. I expected the imported model to be rotated such that the normals are now parallel.

### Actual behavior

The solver failed to converge. If I rotate the imported model at all (in any direction) before applying the constraint, then it works as expected. Thus I suspect it is the fact that the model is imported with its centre point normal exactly perpendicular to the existing centre point normals that is causing the problem.

### Additional information

See `matrix-right.slvs` in the attached file.

phkahler, 2022-01-07:
@henrebotha what about the same orientation constraint?

phkahler, 2022-01-11:
Going with a wont-fix on this one. The issue is mathematical and there isn't an obvious way to push the solver in one direction or the other.

closed 2022-01-11 by phkahler