solvespace/solvespace #1149: Study a possible solver improvement

phkahler, 2021-12-10:
### System information

- **SolveSpace version:** 3.x
- **Operating system:** All

A discussion recently came across [hackernews](https://news.ycombinator.com/item?id=29472189) about an [improved newtons method](https://arxiv.org/abs/2112.02089) with better convergence.  The suggestion is for someone to study that to see if it could improve our solver performance.

That paper obviously isn't the only option for improvements, but it seemed interesting and I thought I'd document its existence here.

zwn, 2024-02-07:
Since gradient descent has received some much work recently (think all the neural networks training), there could be a code ready to be used without the need to too deep into the implementation. The problem the solver is solving here also seems to be similar to [SLAM][1] problems in robotics - the basic idea is that the robot measures different distances and/or angles to different parts of the environment and the goal is to create a globally consistent map of the environment/features. That seems like 1:1 mapping to the constraints solvespace is using. The SLAM implementations also focus on being able to cheaply add a new measurement/constraint and incrementally recompute the problem.

[1]: https://en.wikipedia.org/wiki/Simultaneous_localization_and_mapping

One such a library could be [GTSAM](https://github.com/borglab/gtsam). A comprehensive intro and background for the math can be found in [Factor Graphs and GTSAM](https://gtsam.org/tutorials/intro.html). Another one from a similar domain is [ceres solver](http://ceres-solver.org/). I believe it was at one time used for Street View calculations. Or maybe [g2o](https://openslam-org.github.io/g2o.html).

Why am I interested? Well I think solvespace is a great piece of software but I am suffering from #1354 and maybe also other problems that I have not been able to find yet in the issues.

zwn, 2024-02-09:
The "homepage" of the paper on the improved newtons method is https://www.konstmish.com/publication/21_newton/ and it links a video, an example code in python and other resources.

ruevs, 2024-02-10:
And the code specifically https://github.com/konstmish/opt_methods/blob/master/optmethods/second_order/reg_newton.py