solvespace/solvespace #1150: SolveSpace freeze on enter U+EA00 character for TrueType Text

Symbian9, 2021-12-15:
### System information

- **SolveSpace version:** 3.0~https://github.com/solvespace/solvespace/commit/85f6ec414492d07b512fdfc6f97559ec11ebb36c
- **Operating system:** Debian 10 buster

### Expected behavior

SolveSpace should not be freezed by input any Unicode symbol for TrueType Text.

### Actual behavior

SolveSpace fully freeze after input `` (U+EA00) character into text input field and apply (click <kbd>Enter</kbd>)

1. Launch SolveSpace;
2. Add TrueType Text object (press <kbd>T</kbd>);
3. Select it & input `` into text input field (no matter what font is selected);
4. Press <kbd>Enter</kbd> to apply.

Occasionally I also double clicked on "Property Browser" window title (after SolveSpace frozen) — as a result I got "two" windows of "Property Browser" (all are totally frozen including main window).

[image]

[image]

[image]

### Additional information

FTR, There is Twitter font with the only character symbol is `` (U+EA00).

- https://github.com/ctrlcctrlv/TwitterEA00
  - https://twitter.com/fr_brennan/status/1435780076537274372

BUT, I tested with other fonts (including in-built Bitstream Vera) — SolveSpace always freezes on `` (U+EA00) input, no matter which font is selected.

cc @ctrlcctrlv

phkahler, 2021-12-15:
U+EA00 is in the "private use area" and is not standard unicode. On the other hand, solvespace should not lock up if someone tries to input characters like that - assuming you're after a twitter bird.

ruevs, 2021-12-16:
It's alive!

[image]

I'll submit a pull request when I write up the stupidity in our binary search for glyphs insige unifont.hex.

By the way the font is no good - zero length edge on the "beak".

ctrlcctrlv, 2021-12-16:
All glyphs in TrueType fonts (SFNT quadratic `glyf`) are implied closed. Don't blame the font for errors in rendering implementation please. Due to the fact font authors know there will be an implied force closure, many fonts, not just mine, eliminate an edge to save filesize.

I will fix this font probably because it's so small and this isn't a concern but you should know it.

ruevs, 2021-12-16:
The fifteen or so other true type fonts ("Abc", obviously not all glyphs) that I just tried and all others I've used in the past to create 3D geometry result in closed contours and no zero length edges in SolveSpace. I did try to look through the glyph https://github.com/ctrlcctrlv/TwitterEA00/blob/main/TwitterEA00.ufo/glyphs/uea00.glif but could not quite parse where the problem is.

That said this bug has nothing to do with the Twitter font. SolveSpace is hanging when looking for missing glyphs in its embedded vector font. So I'm certainly not blaming it on the TTF.

phkahler, 2021-12-18:
I updated the title since it's not the True Type font failing. It's the unifont in the text window.

ruevs, 2021-12-20:
In the meantime the twitter font is fixed https://github.com/ctrlcctrlv/TwitterEA00/commit/862c14dd3d05cf4a5a11fc08e7e4d130ca79d857 and now one can have a 3d Twitter :-)

Thanks you @ctrlcctrlv .

[image]

Symbian9, 2021-12-20:
> and now one can have a 3d Twitter

This one also is waiting for DEB package to be built from `master` ater related pull to be merged)

BTW, Thank you all for working on fixing this issue! Can't wait for the next 'nightly' SolveSpace builds for crash testing;)

ctrlcctrlv, 2021-12-21:
You're welcome, funny enough this makes TwitterEA00 font the first font that I've publicly released which is built with my MFEK editor, I fixed the glyph in MFEKglif lol.

closed 2021-12-21 by phkahler

ruevs, 2021-12-21:
> You're welcome, funny enough this makes TwitterEA00 font the first font that I've publicly released which is built with my MFEK editor, I fixed the glyph in MFEKglif lol.

Hey you are the author of MFEKglif! Having never heard of "glif" before your Twitter font; just yesterday I Googled it and found MFEKglif - pretty cool and written in Rust too. Downloaded the Windows binary and "clicked around" a bit with the Twitter font - nice :-)