solvespace/solvespace #1153: snapcraft in CD fails: Error fetching account information from store: The request is missing an Authorization header field containing a valid macaroon

ppd, 2021-12-22:
https://github.com/solvespace/solvespace/runs/4596340741?check_suite_focus=true

Probably just because a new major snapcraft version has been released. I guess the login token has a different format and needs to be exported again. Will look into it very soon.

ppd, 2022-01-02:
It seems that the macaroon has simply expired. I'll send a new one to @phkahler.

phkahler, 2022-01-07:
Deploy worked for both snaps. Assuming the "Upload Release Assets" step is only for tagged release we can close this.

ruevs, 2024-04-17:
@ppd @phkahler it seems the secret has expired again:
https://github.com/solvespace/solvespace/actions/runs/8725534390/job/23939022636

```
Exported credentials are no longer valid for the Snap Store.
Recommended resolution: Run export-login and update SNAPCRAFT_STORE_CREDENTIALS.
For more information, check out: https://snapcraft.io/docs/snapcraft-authentication
Full execution log: '/home/runner/.local/state/snapcraft/log/snapcraft-20240417-165105.684578.log'
```