solvespace/solvespace #1161: Building solvespace as a submodule fails

ToddG, 2021-12-23:
### System information

#### HOST
Linux starbird 5.11.0-41-generic #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

#### GUEST
ubuntu:latest

solvespace:main branch

### Expected behavior

I should be able to build solvespace even if it is a submodule of another repo.

### Actual behavior

I have created a simple docker container to build and run solvespace here: https://github.com/ToddG/solvespace-builder
I wanted to include solvespace as a git submodule and then just recursively update the submodules and build. However, including solvespace as a submodule within my repo does not work b/c cmake fails to get the git hash.

#### REPRO

git clone https://github.com/ToddG/solvespace-builder.git
git co toddg/submodule-repro-error

You'll need a docker build container (this is just my workflow, the error would show w/o docker involved)
```
$ ./build-container.sh
```

Now build using the submodule:
```
$ ./compile-solvespace.sh 
access control disabled, clients can connect from any host
CMake Deprecation Warning at CMakeLists.txt:14 (cmake_policy):
  The OLD behavior for policy CMP0048 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Error at cmake/GetGitCommitHash.cmake:8 (file):
  file failed to open for reading (Not a directory):

    /mnt/source/.git/HEAD
Call Stack (most recent call first):
  cmake/GetGitCommitHash.cmake:35 (get_git_commit_hash)
  CMakeLists.txt:38 (include)

-- found OpenMP, compiling with flags: -fopenmp
-- Using in-tree libdxfrw
-- Using in-tree mimalloc
-- 
-- Library base name: mimalloc
-- Build type       : release
-- Compiler         : /usr/bin/cc
-- Install directory: /usr/local/lib/mimalloc-2.0
-- Build targets    : static
-- 
CMake Warning at CMakeLists.txt:324 (message):
  Gettext not found, translations will not be updated

-- Configuring incomplete, errors occurred!
See also "/mnt/build/CMakeFiles/CMakeOutput.log".
```

This is failing b/c there is no `.git` directory in a git submodule:

```
$ ls -lsa solvespace-source
total 172
 4 drwxrwxr-x 13 toddg toddg  4096 Dec 23 13:41 .
 4 drwxrwxr-x  8 toddg toddg  4096 Dec 23 13:41 ..
 4 drwxrwxr-x  2 toddg toddg  4096 Dec 23 13:41 bench
16 -rw-rw-r--  1 toddg toddg 14921 Dec 23 13:41 CHANGELOG.md
 4 -rw-rw-r--  1 toddg toddg  1241 Dec 23 13:41 .clang-format
 4 drwxrwxr-x  2 toddg toddg  4096 Dec 23 13:41 cmake
16 -rw-rw-r--  1 toddg toddg 14367 Dec 23 13:41 CMakeLists.txt
12 -rw-rw-r--  1 toddg toddg 11204 Dec 23 13:41 CONTRIBUTING.md
36 -rw-rw-r--  1 toddg toddg 35148 Dec 23 13:41 COPYING.txt
 4 drwxrwxr-x  2 toddg toddg  4096 Dec 23 13:41 developer_docs
 4 drwxrwxr-x  2 toddg toddg  4096 Dec 23 13:41 exposed
 4 drwxrwxr-x 11 toddg toddg  4096 Dec 23 13:41 extlib
 4 -rw-rw-r--  1 toddg toddg    42 Dec 23 13:41 .git
 4 -rw-rw-r--  1 toddg toddg   132 Dec 23 13:41 .gitattributes
 4 drwxrwxr-x  4 toddg toddg  4096 Dec 23 13:41 .github
 4 -rw-rw-r--  1 toddg toddg   235 Dec 23 13:41 .gitignore
 4 -rw-rw-r--  1 toddg toddg   789 Dec 23 13:41 .gitmodules
 4 drwxrwxr-x  2 toddg toddg  4096 Dec 23 13:41 include
 4 drwxrwxr-x  4 toddg toddg  4096 Dec 23 13:41 pkg
12 -rw-rw-r--  1 toddg toddg  9340 Dec 23 13:41 README.md
 4 drwxrwxr-x 10 toddg toddg  4096 Dec 23 13:41 res
 4 drwxrwxr-x  5 toddg toddg  4096 Dec 23 13:41 src
 4 drwxrwxr-x  7 toddg toddg  4096 Dec 23 13:41 test
 4 -rw-rw-r--  1 toddg toddg  2873 Dec 23 13:41 THIRD_PARTIES.txt
 4 -rw-rw-r--  1 toddg toddg   510 Dec 23 13:41 wishlist.txt
```

### Additional information

For now, my work-around is to just clone the solvespace repo into my repo manually.

See: 

```
## so here we manually clone if it doesn't already exist
#if [[ ! -d solvespace-source ]]
#then
#    git clone https://github.com/solvespace/solvespace solvespace-source
#fi
```

But ideally you'd fix your cmake command or push a bug up to cmake?

ruevs, 2021-12-23:
https://github.com/solvespace/solvespace/blob/71c6492d6de798166e0c09d1529dffed745f6fbf/CMakeLists.txt#L35

rpavlik, 2022-01-02:
so this is almost certainly a CMAKE_SOURCE_DIR vs CMAKE_CURRENT_SOURCE_DIR or PROJECT_SOURCE_DIR or ... thing. My recent build improvements may have helped.

phkahler, 2022-06-02:
@ToddG  Is this any better?

ruevs, 2024-08-18:
No one cares it seems. Closing.

closed 2024-08-18 by ruevs