solvespace/solvespace #1164: Add "osifont" as default built-in TrueType font

Symbian9, 2021-12-25:
### System information

- **SolveSpace version:** `3.x`, `4.x`
- **Operating system:** all supported

### Expected behavior

Replace actually built-in TrueType font (which does not related to CAD) with ***"osifont"*** which is conforms `IS0 3O98` standard as font for CAD & engineering/technical drawings:

- Source & TTF binary: https://github.com/hikikomori82/osifont
- Online font preview (specimen): https://hikikomori82.github.io/

### Actual behavior

Actually built-in font in SolveSpace is *"Bitstream Vera Sans (Roman)"* which is not CAD-specific font at all.

### Additional information

[image]

ruevs, 2021-12-25:
"2021-12-07 - fixed kerning in P.(A-Za-z), removed Belarusian support in protest of latest actions on Polish border" now there is something one does not see often - a CAD font with a political agenda :-)

Anyway can you explain why should the default font be a CAD font? I would think that the usage of a font would be to add text to a 3D model - it's highly unlikely whoever is doing it will want the text to be in a mechanical drawing ISO font.

But the font is nice to know about. For example for making ISO conforming drawings.

Symbian9, 2021-12-25:
> Anyway can you explain why should the default font be a CAD font?

First of all, as for TrueType font - in most cases used for extrude, so the best choice for such font is a font with same width along each character path ("isofont" is exactly what it should look like, but not "Bitstream Vera Sans")

Few other fonts I may recommend to look:

- "OpenGost" - set of OFL-licensed fonts which conforms exUSSR's GOST 2.304-81 standard and its derivatives still used by post-soviet states: 
  - ~~https://bitbucket.org/fat_angel/opengostfont~~
  - https://github.com/aur-archive/opengostfont
    - https://packages.altlinux.org/en/sisyphus/srpms/opengostfont/

"OpenGost" fonts also converted into LFF-format (as outlined fonts) and bundled with LibreCAD:

- https://github.com/LibreCAD/LibreCAD/tree/master/librecad/support/fonts
  - "OpenGost Type A (Regular)" - https://github.com/LibreCAD/LibreCAD/blob/master/librecad/support/fonts/OpenGostTypeA-Regular.lff
  - "OpenGost Type B (Regular)" - https://github.com/LibreCAD/LibreCAD/blob/master/librecad/support/fonts/OpenGostTypeB-Regular.lff

ruevs, 2021-12-26:
> First of all, as for TrueType font - in most cases used for extrude, so the best choice for such font is a font with same width along each character path

The ISO/ГОСТ/DIN etc. technical drawing fonts are constant stroke width, because they were drawn with [technical pens (rapidographs)](https://en.wikipedia.org/wiki/Technical_pen) and I agree that for extruding and injection moulding (of embossed letters) it is an advantage. However this is only one of many possible uses - for engraved (debossed) letters, CNC, 3d printing, cutting (laser, water jet, plasma, vinyl...) it does not matter and obviously designers will schoose a font based on aesthetic considerations.

Because of the above and backwads compatibility I do not think it is a good idea to change Vara Sans. And even adding osifont as a second embedded one goes against SolveSpaces' minimalism.

However I do like having a reference for these fonts so I'll link this in the "FAQ" issue #1024.

Symbian9, 2021-12-26:
> However I do like having a reference for these fonts so I'll link this in the "FAQ" issue `#1024`

Ok, then I will close this issue for noww in favor to mentioning them in "FAQ"

FTR, @phakhler, may you enable "GitHub Wiki" feature for this repo?

closed 2021-12-26 by Symbian9