solvespace/solvespace #1166: README : xcode build instruction update

brennane, 2021-12-26:
### System information

- **SolveSpace version:** .N7tptBcGk3$$K
- **Operating system:** macos 12.1

### Expected behavior

xcode project build should work.

### Actual behavior

Following the README file on current macos is failing. 

1. the workaround (for README inclusion) is simple, see additional information
2. the cmake process should be updated for the new build system on mac

~~~
% cmake .. -G Xcode
CMake Deprecation Warning at CMakeLists.txt:14
. . .
-- The C compiler identification is AppleClang 13.0.0.13000029
-- The CXX compiler identification is AppleClang 13.0.0.13000029
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang
. . . 
-- Generating done
CMake Error in extlib/libpng/CMakeLists.txt:
  The custom command generating

    /Users/breevans/Work/Wumpus/solvespace/build/extlib/libpng/pnglibconf.c

  is attached to multiple targets:

    genvers
    gensym
    genfiles

  but none of these is a common dependency of the other(s).  This is not
  allowed by the Xcode "new build system".

. . . 

CMake Generate step failed.  Build files cannot be regenerated correctly.
~~~

### Additional information

Flag `-T buildsystem=1` to force the old build behavior works:

~~~
cmake .. -G Xcode -T buildsystem=1
. . .
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   zlib

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: [private]/solvespace/build
~~~

And now the project will build just fine

ruevs, 2022-01-24:
`cmake .. -G Xcode` worked just fine for me on it's own, see #1199 but that is probably because I used macOS 10.14.6 and Xcode 11.3.1.

On the other hand `cmake .. -G Xcode -T buildsystem=1` worked as well, but the application size (in debug mode) is 1144 bytes smaller (47882062 vs. 47880918),..

This seems to be a problem with the `libpng` CMake configuration. Am I correct @vespakoen, @rpavlik ?

rpavlik, 2022-01-29:
I have no idea about anything mac anymore, sorry.

phkahler, 2022-02-04:
@vespakoen  thoughts?

phkahler, 2022-04-19:
Pinging @vespakoen again ;-)

vespakoen, 2022-04-19:
This probably has to do with the cmake version, @brennane what version of cmake are you using?

I have already opened up an issue over at libpng to update the targets, but it seems to take a while for them to change it.

I think it makes sense to add the extra flag to our docs.

brennane, 2022-05-10:
MacOS:  Confirmed that as of `cmake version 3.23.0` (from homebrew) solvespace is building without the backwards compatability flag.  I can't say that it was the cmake change or a dependent library update; regardless Solvespace certainly builds with no extra flags at this point.

closed 2022-05-10 by brennane