solvespace/solvespace #1168: Solvespace Windows are completely stuck after attempting to change the Background Colour

ConductiveInsulation, 2021-12-28:
### System Info:

- **SolveSpace version:** 3.0~2450010b
- **Operating system:** Linux Mint 20.2 Cinnamon
- **Kernel** 5.4.0-91-generic
- **CPU** Intel© Core™ i5-8265U CPU @ 1.60GHz × 4
- **RAM** 7.9gb free of 15.1 

### Expected behavior

Change of BG Colour since it was hard to Distinguish Background and a Black Object in the file. 

### Actual behavior

PID28419 Was the Window opened when i decided to change the Background colour. Could change it a few times but Scrolling between the Colours didnt work (Still doesnt for newly opened windows). Noticed the Window got stuck. SInce the File was a new, unsaved File so I cant upload a copy since only the version in the Ram exists, maybe its even in the Memory Dump i created from [This snippet](https://serverfault.com/a/408929)

The second Window had the same File as in PID38019 Open, Was possible to save and close it. Unfortunately i forgot to save informations about it.

PID32815 was opened later to try if i can replicate it for the Issue here. Currently stuck too, mostly empty. 

### Additional informations to the Processes
[PID38019.txt, opened later, works](https://github.com/solvespace/solvespace/files/7784975/PID38019.txt)
[PID32815.txt, Opened later, stuck](https://github.com/solvespace/solvespace/files/7784965/PID32815.txt)
[PID28419.txt, Open when issue occured first](https://github.com/solvespace/solvespace/files/7784966/PID28419.txt)
[mem.tar.gz, memory of 28419](https://github.com/solvespace/solvespace/files/7785062/mem.tar.gz)

Ill Try to keep the Process open as long as possible to maybe get further Informations, since Im not a Programmer I'm not familiar with whats needed to maybe solve that. Just tell me what might be helpful.

phkahler, 2021-12-30:
@ConductiveInsulation  I tried to reproduce this. One thing I did see is that once I click "change" under background color, the main window becomes completely unresponsive until a color is selected and the color chooser goes away. Then everything is fine again. Maybe that's what you were seeing?

Symbian9, 2021-12-31:
> Maybe that's what you were seeing?

@ConductiveInsulation As an addition, try to repeat all the steps for reproduce background color change BUT then just try click <kbd>>Esc</kbd> key on keyboard (once or twice).

ConductiveInsulation, 2021-12-31:
@Symbian9 @phkahler 

I tested a few things. I could not reproduce another window being affected but I could completely get the current window stuck
Pressing `esc` only works as long as the program isnt completely stuck
###How to get current window stuck

1. Open Dialog to change BG colour
2. Click change to view colour selection
3. dont select colour
4. close Dialog
5. Do literally anything except pressing enter
6. press enter a few times
     depending what you pressed you have to confirm a dialog before Solvespace closes itself

###Conclusion
I think i remembered a few things false while writing the Issue, I could not manage to get a instance of SolveSpace stuck from another instance.  As long as i dont find out what I did i think its save to assume i just reproduced it on multiple Windows and forgot it.

###Possible fix
When dialog gets closed end the mode that freezes everything.

phkahler, 2022-01-01:
@ConductiveInsulation  How do you close a dialog in step 4?  I'm on linux (Fedora) and the color chooser isn't a separate dialog, it's part of the Property Browser window.

Also, are you on X or Wayland?

ConductiveInsulation, 2022-01-01:
@phkahler With dialog i meant the Property Browser. Updated the previous Comment with a comment. Has to get closed with the X, tab doesnt work 

I'm directly on my Notebook, no Protocol in between. If you want I can make a Screen recording for you.

Symbian9, 2022-01-01:
> If you want I can make a Screen recording for you.

It would be the best to understand.

You may attach MP4 files directly to your comments here.

ConductiveInsulation, 2022-01-02:
@Symbian9 
Only used the enter key in the end after it got stuck
https://user-images.githubusercontent.com/85430982/147884380-2c0eb0ee-bcc8-4b23-b2d8-ddae9035082a.mp4

Symbian9, 2022-01-02:
> Only used the enter key in the end after it got stuck

OK, this bug is reproducible.

@phkahler, it seems like "Property Browser" window should be locked (prevents closing it) during any input box is activated.

Another possible solution to fix this bug: when "Property Browser" window is closed manually during any input field is activated SolveSpace should firstly close input field process.

ConductiveInsulation, 2022-01-02:
Guess I should have made the video in the beginning. But I'm still learning, usually I don't break software.

ruevs, 2022-01-06:
@ConductiveInsulation if you can build from source can you check if this https://github.com/solvespace/solvespace/pull/1180 fixes the problem on Linux?

On Windows the hang does not happen but there is still a subtle bug.

ruevs, 2022-01-06:
I found one more bug related to the color picker - it remains open when hitting `ESC`  - but that is a bit trickier to fix.

closed 2022-01-06 by phkahler

ConductiveInsulation, 2022-01-07:
@phkahler sorry for the late reply, tested it and it works as it should.

PR #1180 (2022-01-06): Hide edit boxes before closing the property browser.
When the text window/property browser is closed while and edit box
is active its window remained open. This is incorrect and probably
causes a hang on Linux described in #1168. So hide the edit control
when closing.

Probably Fixes #1168