solvespace/solvespace #1169: [feature request] export several STL files from a single slvs file

bmorel, 2021-12-29:
This is a feature request, which would help multi-material 3D printing a lot.

Currently, when one wants to 3D print with multiple materials, it is required to do each drawing separately, and to *manually offset the 1st layers of each* which is error-prone and cumbersome.

What I propose, and I'm ok to implement it myself if I can have some guidance on how to do it, is to be able to export stl files based on volumes. From what I understand, this basically includes extrusions and all transformations which applies to those.

For example, to print this:
[image]

I needed to generate 2 stl files, which required me to find a trick* (actually, I used several tricks to quickly generate the full set of dominoes, including shell scripting, because I'm lazy) to offset the inside of the dots enough so that the FOSS program I use to slice them (prusa-slic3r, a fork of slic3r) could group those stl files into a single multi-material file.

*: The trick is to extrude an empty schema, and to remember to modify all files to match. Quite annoying to do, trust me, and I would not dare try this for complex parts

bmorel, 2021-12-29:
Actually, a better way would be to export a single AMF/3MF file, since they are built for that kind of feature, but I think exporting several stl files would be a good, easier to implement, 1st step.

you can find infos here: https://blog.prusaprinters.org/3mf-file-format-and-why-its-great_30986/ and I have to say that the few exchanges I had with prusa persons were rather nice, so maybe they'd help. They do, after all, release all their stuff under FOSS licences (and it would be commercially interesting for them to have free&gratis 3D drawing software supporting that).

ruevs, 2021-12-29:
Would it not be easier to draw the separate volumes as separate SolveSpace files and then link/assemble them in a new file? 

Here is a tutorial https://solvespace.com/box.pl
And the.one before last video at the bottom here: https://solvespace.com/tutorial.pl

phkahler, 2021-12-30:
I'm not excited about 3mf, but if you want to write an explorer that would be OK. We already have a request for glTF which is probably better, and also not a priority.

The main issue seems to be how to produce variants of a given part/sketch. Would it be possible to achieve what you need by checking "suppress solid model" on certain groups? What about the use of color? We do support color in .stl files - both writing and linking.

The key question it seems isn't so much the file formats as how to use solvespace to specify what you want. Once we figure that out the rest should follow?

bmorel, 2021-12-31:
Yes, the feature request is more about being able to export parts of the 3D model than to support new file formats, which are of dubious interest considering the work they'll likely require (possibly high cost for minor benefit, I would not call that worthwhile).

I do not know if it's already possible with checking "suppress solid model", but, to me, the generated stl file does not checks the state of what is shown on screen.