solvespace/solvespace #1171: Can't draw arc

jelinj8, 2022-01-01:
### System information

- **SolveSpace version:** 3.0~0e0b0252
- **Operating system:** Win 10 x64 21H2

### Expected behavior

select ARC tool, mark first endpoint, mark second endpoint.

### Actual behavior

select ARC tool, first click creates "unsolvable constraints" (with empy list of "unsatisfied constraints").
Endpoints and center are created on one point as if I've set both endpoints.

Doesn't matter if it is a new file or existing drawing.

### Additional information

It is possible to move the points later (sometimes center as first, sometimes one of endpoints). Then it works, just the initial drawing is bad.

phkahler, 2022-01-01:
To be clear:

1) Start Solvespace
2) Click the Arc tool
3) Click somewhere on the sketch not on the origin.
4) Problem?

And this is with Version 3.0 release on Windows?

jelinj8, 2022-01-01:
[image]

phkahler, 2022-01-01:
Did it ever work for you? On that computer?

jelinj8, 2022-01-01:
Hmm.. interresting question. I think that it used to work as I was using Solvespace quite a lot for 3D printing, but I've tried an older version of Solvespace now and that doesn't work either. I've tried the new version on another computer and it works. It isn't a problem of mouse doubleclicking (I've tested another mouse), neither multiscreen issue. Only thing I can't test is another computer with dedicated graphics card (nVidia 1050Ti) as I don't have any (only notebooks). I haven't tested it outside my main work PC previously, now it seems to be an isolated issue on this one.

I'm a software developer, but I have no idea what else to try, except reinstalling the workstation :-(

Sorry for bad testing.

Symbian9, 2022-01-01:
I guess I can reproduce this issue: it seems like during new "Arc" creation you clicked <kbd>LMB</kbd> second time too fast in a row after first time (may be that is mouse issue) - as a result *start point* of "Arc" collided with its *end point*.

https://user-images.githubusercontent.com/13188580/147860720-bc7484e1-d4c3-4fb5-95c9-6467a5324793.mp4

Just as recommendation, always try to draw bigger arc with carefully to not click second time after first time in same/very close place on canvas.

@phkahler, may be it would be useful to hardcode some pause/delay time between first and second mouse keys & keyboard clicks?
(there was already similar issue thread on issues tracker, if I will find it will link it here too)

jelinj8, 2022-01-01:
The thing is that I'm not clicking at all for the second time and for the mouse issue it'd have to be some system issue as it is the same with several mouses. Plus it is an issue only with arc with no relations, if I start from a point or try to draw anything else than arc, there is no problem.

phkahler, 2022-01-02:
@jelinj8  Like @Symbian9 says, I can reproduce this (on Linux) if I double click the first point of an arc. The funny thing is that it creates the entire arc, and I can drag the 3 points apart one at a time.  If I draw line segments (s) and double click the first point I get a full segment there along with a pts-coincident constraint.

Some googling indicates that some Windows 10 users are experiencing mouse double-clicking when they only click once. For example, selecting a file in explorer with a single click opens it instead, which is double-click behavior. Apparently this may have been due to a Windows update?

https://www.google.com/search?q=windows+10+double+click+bug

If that's the issue, I want Microsoft to give me Holiday pay for this one ;-)

Symbian9, 2022-01-02:
> If that's the issue, I want Microsoft to give me Holiday pay for this one ;-)

Fully agree! There should be NO fix for Windows without Bounty-pay from MS for SolveSpace Team ;P

Happy Holidays, Y'all! :)

ruevs, 2022-01-06:
Well this is not a bug in SolveSpace strictly speaking. Clicking twice on the same spot does create a "degenerate" ark that is unsolvable, but is can still be "dragged apart" to normal with a few attempts at the three points stacked on top of each other.

As for the potential double click problem in Windows - here is one decent looking reference https://www.computerhope.com/issues/ch001532.htm

I will close this, @jelinj8 if you think that the problem you see is something different feel free to describe it here.

closed 2022-01-06 by ruevs