solvespace/solvespace #1173: Snapcraft build issue

vespakoen, 2022-01-02:
### System information

Ubuntu 20.04.3 (in Github Actions)

### Expected behavior

Snap should build successfully

### Actual behavior

Fails with ld linking errors

### Additional information

Working fine here: https://github.com/solvespace/solvespace/actions/runs/1485714200
Broke in the next push to master: https://github.com/solvespace/solvespace/actions/runs/1564051582

The commits in between are: https://github.com/solvespace/solvespace/compare/85f6ec4..91db627 and are unrelated / don't touch any snap stuff.

So I guess the issue is in the environment.
Both runs seem to use Snapcraft 5.0, the Ubuntu version is also the same, although the Github Actions image changed a little.

I tried fixing the missing cairo library with this commit: https://github.com/solvespace/solvespace/commit/b71c7282622b74f3f74a33d7801357d95f8a4d2e
 (here is the action run for it: https://github.com/solvespace/solvespace/actions/runs/1646157739)

Which made the missing -lcairo go away, but presents other missing libraries.

So my guess is that some library path is not provided anymore or is incorrect.

I don't have a linux computer setup at the moment, so I can only test things by going through CI, which is a bit annoying.

Hoping @ppd can help here ;)

ppd, 2022-01-02:
> I tried fixing the missing cairo library with this commit: b71c728

I'll take a look later today. Should be easy to fix.

ppd, 2022-01-02:
I'm very confused. CMake seems to use ld as the linker, which ignores LIBRARY_PATH. I thought the default was to use g++ for linking.

PR #1174 (2022-01-02): CMake Fixes + Snap port to core20
In 006539b, solvespace_MAJOR_VERSION etc. were removed.
However, these variables were still referenced in some places.
Solution: Use PROJECT_VERSION instead.

006539b removed the call to link_directories for gtkmm, jsonc & fontconfig.
This leads to linking errors if those libraries are in "non-standard"
paths.
Fix this by introducing a target specific target_link_directories call.

Fixes #1173

Moving to core20 was long overdue anyway, and
the recent CMake changes necessitated some fixes.
Also switch to LZO compression for (way) better cold start
performance.
 [... cut]
commit 34efb6de (2022-01-02): CMake Fixes + Snap port to core20 (#1174)

* CMake: use PROJECT_VERSION instead of solvespace_*_VERSION

In 006539b, solvespace_MAJOR_VERSION etc. were removed.
However, these variables were still referenced in some places.
Solution: Use PROJECT_VERSION instead.

* CMake: re-add link directories for solvespace target

006539b removed the call to link_directories for gtkmm, jsonc & fontconfig.
This leads to linking errors if those libraries are in "non-standard"
paths.
Fix this by introducing a target specific target_link_directories call.

Fixes #1173

* snap: port to core20 & adapt to CMake c [... cut]
closed 2022-01-02 by vespakoen