solvespace/solvespace #1177: Rename "line styles" to "styles"

Symbian9, 2022-01-03:
### System information

- **SolveSpace version:** `3.0~`https://github.com/solvespace/solvespace/commit/e07b082eb810d2088c92b97d04e864a4ab95816f
- **Operating system:** all supported

### Expected behavior

In "Property Browser" tab "line styles" should be renamed just to "styles".

### Actual behavior

Actually "line styles" are misinterpreted as only styles for lines, but it also for assigns not only for lines, but also for closed path fills & solid colors.

### Additional information

phkahler, 2022-01-03:
"styles and colors" ???

Symbian9, 2022-01-03:
> "styles and colors" ???

Maybe. But if so then "colors & styles" (and use "&" instead of "and" for shorten)

Additionally, what You think about move "background color" to "view" tab?

phkahler, 2022-01-07:
> Additionally, what You think about move "background color" to "view" tab?

In that case this should remain "line styles" because it's just for lines?  I did wonder about moving BG color, but if the existing tab if for colors then no?

Question: Are any of these saved as part of the sketch, or are they all general settings?

ruevs, 2022-01-07:
@phkahler if you create custom styles they are shown in the list and are saved in the model. The default ones are in the registry/json.

Changing label is superfluous (especially considering the translations).

closed 2022-01-07 by ruevs

Symbian9, 2022-01-07:
> Question: Are any of these saved as part of the sketch, or are they all general settings

Good question!

Guess, "background color" is not saved into `slvs` file, but other "line styles" (especially newly created custom styles) are saved into `slvs`, where there are entities with custom styles.