solvespace/solvespace #1186: Performance testing models (slow speed benchmarks)

ruevs, 2022-01-09:
This is a list of SolveSpace models that have shown performance bottlenecks in the past. Some of them were resolved, some were improved but still are slow.

I have marked all related issues and pull requests (that I could find or remember) with the [`performance`](https://github.com/solvespace/solvespace/issues?q=label%3Aperformance+sort%3Acreated-desc) label.

Links to the `.slvs` models reproducing the problems and the fixes may be useful for performance testing.
If the "Fixed on" column is empty the issue is probably still open (or I did not update this list :-)

Source | Model | Author | Fixed on | Fixed in
------ | ----- | ------ | -------- | --------
https://github.com/solvespace/solvespace/issues/131 <br/> https://github.com/solvespace/solvespace/issues/225 | [constrainfreeze.zip](https://github.com/solvespace/solvespace/files/645378/constrainfreeze.zip) | @whitequark | 2020-09-16 workaround | https://github.com/solvespace/solvespace/commit/615708440f97570cd7055ec65ec0a3bbe6f8f18d
https://github.com/solvespace/solvespace/issues/386#issuecomment-544301008 <br/> https://github.com/solvespace/solvespace/issues/386#issuecomment-998587512 <br/> https://github.com/solvespace/solvespace/issues/386#issuecomment-1000371505https://github.com/solvespace/solvespace/issues/386#issuecomment-1000371505 <br/>  | [profiling.zip](https://github.com/solvespace/solvespace/files/3748504/profiling.zip)<br/>[keyb_case_MAX_UNKNOWNS_2048.zip](https://github.com/solvespace/solvespace/files/7750846/keyb_case_MAX_UNKNOWNS_2048.zip)<br/>[keyb_case_MAX_UNKNOWNS_4096.zip](https://github.com/solvespace/solvespace/files/7770072/keyb_case_MAX_UNKNOWNS_4096.zip)<br/>[]()| @bmorel<br/>@ruevs<br/>@ruevs<br/> | 2021-12-23 | https://github.com/solvespace/solvespace/pull/1159
https://github.com/solvespace/solvespace/pull/594#issuecomment-624983188 <br/>  <br/>  <br/>  | [intersect_ct1.zip](https://github.com/solvespace/solvespace/files/4590263/intersect_ct1.zip) | @phkahler | 2020-05-08 | https://github.com/solvespace/solvespace/pull/594
https://github.com/solvespace/solvespace/pull/616#issuecomment-632057911 <br/> https://github.com/solvespace/solvespace/pull/616#issuecomment-632062869 | [LatheSplineIntersectCylinder.zip](https://github.com/solvespace/solvespace/files/4662392/LatheSplineIntersectCylinder.zip)<br/>[A_wheel.zip](https://github.com/solvespace/solvespace/files/4662425/A_wheel.zip) | @ruevs<br/>@phkahler | 2020-05-23 | https://github.com/solvespace/solvespace/pull/616
https://github.com/solvespace/solvespace/issues/759 <br/> https://github.com/solvespace/solvespace/issues/759#issuecomment-713442468 | [perf_test.zip](https://github.com/solvespace/solvespace/files/5412461/perf_test.zip)<br/>[PrismConeNURBSNormalsTangents300.zip](https://github.com/solvespace/solvespace/files/5414683/PrismConeNURBSNormalsTangents300.zip) | @phkahler<br/>@ruevs | 2020-10-23<br/>2020-10-24<br/>2020-10-26 | https://github.com/solvespace/solvespace/pull/765<br/>https://github.com/solvespace/solvespace/pull/762<br/>https://github.com/solvespace/solvespace/pull/780
https://github.com/solvespace/solvespace/issues/841#issuecomment-742153838 <br/> https://github.com/solvespace/solvespace/issues/932 <br/> https://github.com/solvespace/solvespace/issues/932#issuecomment-778506906 <br/>  | [8040_finished_constrained.zip](https://github.com/solvespace/solvespace/files/5669510/8040_finished_constrained.zip)<br/>[case-vents.slvs.gz](https://github.com/solvespace/solvespace/files/5962704/case-vents.slvs.gz)<br/>[LeliaManya2.zip](https://github.com/solvespace/solvespace/files/5974674/LeliaManya2.zip)<br/>[]()| @ruevs<br/>@baryluk<br/>@ruevs<br/> | 2021-03-27<br/>2021-05-13 | https://github.com/solvespace/solvespace/pull/973 https://github.com/solvespace/solvespace/pull/992<br/>https://github.com/solvespace/solvespace/issues/939#issuecomment-810641820 https://github.com/solvespace/solvespace/pull/1002
 <br/>  <br/>  <br/>  | []()<br/>[]()<br/>[]()<br/>[]()| @<br/>@<br/>@<br/> |  |

rpavlik, 2022-01-10:
Do we have copyrights/licenses for each of those so we can include them in a testing repo?

phkahler, 2022-01-10:
A_wheel is probably obsolete for perf testing these days.  Also, if perf_test is the cylinder with many holes, that was posted to the forum so while technically they put it out in public, we probably don't have proper permission to redistribute it as a package.

bmorel, 2022-01-10:
The profiling thing I did is under CC-0. Or WTFPL if you prefer it.

ruevs, 2022-01-10:
Public domain for all of mine.

@rpavlik can you point me to your models please? I did not find any. It's possible I've missed some others too.

Symbian9, 2022-01-10:
> if that was posted to the forum so while technically they put it out in public, we probably don't have proper permission to redistribute it as a package.

To be clear on this and on @rpavlik's note on licensing, even some posts something publicly it does NOT mean that such type of file/content is free to distribute.

Otherwise there should be warning message on forum:

- "Uploading file as an asset to forum you are agreed to allow reuse/redistribute your asset under terms of CC0/CC-BY/CC-BY-SA"

Who knows where file originally may came from.

> Looks like none of mine made the list, but if you find one of mine, CC-BY-4.0

As for my publicly uploaded assets to GitHub & Patreon, I ask to use it under terms of CC-BY-SA-4.0:

- https://github.com/Symbian9/SolveSpace-Daily-Engineering

baryluk, 2022-01-15:
You can take my example model `case-vents.slvs.gz` (any any other variants of it posted on this github issue tracker)  under CC-BY-4.0.

rpavlik, 2022-01-16:
Thanks folks for chiming in with your terms!
I'd recommend using the conventions of https://reuse.software for the benchmarking repo whenever it gets made: license sidecar files for the models is probably the best we can do without adding a "comment-looking" file header to Solvespace files. We can then use their tool in CI to make sure that any models added also have their license and copyright included. The reuse tool is really handy for these purposes, for repos where people might forget to mark files, there are a wide variety of licenses, etc. (Though honestly I tend to make all my projects reuse compliant these days, I'm a big fan, especially for things I might have to make Debian packages for)

ruevs, 2022-01-20:
@rpavlik I did find one of yours https://github.com/whitequark/solvespace-testcases - I had almost forgotten about that collection :-)

hinell, 2022-02-08:
```
Vector::WithMagnitude(1) of zero vector! 
Generate::REGEN (for bounding box) took 15720 ms
Generate::REGEN took 31897 ms
Generate::ALL (for bounding box) took 16577 ms
Generate::ALL took 33203 ms
Generate::DIRTY (for bounding box) took 15852 ms
Generate::DIRTY took 32006 ms
Generate::ALL (for bounding box) took 16113 ms
Generate::ALL took 32466 ms
```

Not sure if I'm in the right spot but I'm having a performance issues with a simple gear model. The culprit is a rotation group. If I rotate a complex sketch over 16 times it gets slower and slower. It takes a couple of minutes to just reopen.

Should I fill a separate issue for this?  The short log is above.  The `Vector::WithMagnitude(1) of zero vector!` string is repeated after every line many times and omitted.

## System
- **Operating system:** Linux Kubuntu
- **SolveSpace version:** 3.0.rc2+repack1-3

## Image & file
* [gear-32-2.zip](https://github.com/solvespace/solvespace/files/8019594/gear-32-2.zip)
[image]

ruevs, 2022-02-08:
@hinell try an edge build.
https://github.com/solvespace/solvespace#via-automated-edge-builds

Is it still slow?

Can we use your model in a potential future benchmark? Under what license?

ruevs, 2022-02-08:
On current [master](https://github.com/solvespace/solvespace/commit/d8b5281f) changing the repeat from 1 to **_64_** in `g00b-rotate` takes `Generate::DIRTY took 2871 ms` ;-)

[gear-64-2.zip](https://github.com/solvespace/solvespace/files/8022949/gear-64-2.zip)

[image]

By the way there are better ways to draw the teeth of the gear ;-)
https://solvespace.com/forum.pl?action=viewthread&parent=3911&tt=1633767543
https://solvespace.com/forum.pl?action=viewthread&parent=2477&tt=1541434762
https://solvespace.com/forum.pl?action=viewthread&parent=916&tt=1439146696
https://solvespace.com/forum.pl?action=viewthread&parent=291&tt=1413300884
https://github.com/solvespace/solvespace/issues/550#issuecomment-655814031
etc.

Symbian9, 2022-02-08:
> On current `master` changing the repeat from `1` to `64` in `g00b-rotate` takes `Generate::DIRTY took 2871 ms`

@ruevs, may you retry performance with switching to *"Hide edges of solid model"*?

hinell, 2022-02-08:
@ruevs 
@Symbian9
Hi. Can I ask you to move all these comments to the related issue?  I was just going to remove my comment here but seems like I forgot to that. Thanks!

ruevs, 2022-02-08:
I removed mine and @Symbian9 should just delete his because the performance with "Hide edges of solid model" is practically the same `Generate::DIRTY took 2723 ms` - this model is solver limited. Not OpenGL, triangulation etc...

For the same reason testing with an OpenGL1.x version is pointless.

Evil-Spirit, 2022-09-29:
[The house built by programmer ](https://github.com/Evil-Spirit/TheHouseBuiltByProgrammer)

[Input.slvs](https://github.com/Evil-Spirit/TheHouseBuiltByProgrammer/blob/main/input.slvs)
[kladka2.slvs](https://github.com/Evil-Spirit/TheHouseBuiltByProgrammer/blob/main/kladka2.slvs)

nazar-pc, 2022-12-24:
After a few years I have finally open sourced my PC case model, which is VERY slow: https://github.com/nazar-pc/NPW-1.0

It has many components, if you open `case.slvs` you'll feel how slow it is (it is much faster now than 3 years ago though) and if you try to modify something, you'll probably start feeling pain.

I hope it is a helpful benchmark of what crazy people might end up using SolveSpace for.

ruevs, 2022-12-27:
1. @nazar-pc that is impressive! Thank you for sharing it. In the meantime, seeing your case someone posted this https://www.thermaltake.com/core-w200.html on Hacker News.

ruevs, 2023-01-06:
@nazar-pc today I tried your fully assembled case. Om [my machine](https://ark.intel.com/content/www/us/en/ark/products/130409/intel-core-i78809g-processor-with-radeon-rx-vega-m-gh-graphics-8m-cache-up-to-4-20-ghz.html) changes (in other words re-generating the model) takes about 6 seconds. This would have been totally infeasible to work with two years ago.

This will be a very useful test model for anyone that decides to tackle performance in the future.

nazar-pc, 2023-01-07:
Absolutely, performance improved A LOT since 2018-2019 when I worked on the model!
I believe I created an assembly relatively earlier in the process so I only needed to update it on changes as individual parts evolved and hide most of the things when modifying to get solve/render time down to something that worked acceptably.

I think a specific case of updating when some things change might benefit from something like https://salsa-rs.github.io/salsa/ so that unnecessary recomputation can be optimized away (I mostly do Rust these days).

ruevs, 2023-01-07:
I pofiled while loading your model. A quick glance at the results shows that 30% of the time is spent searching for triangle edges in a K-d tree. And 10 of those 30% is spent calculating vector magnitudes (square roots). From one side this is good - there are is no glaring slowness in pieces of the code not directly related to geometry - we removed most of those in 2021. On the other hand this means that further speedups will take more work/thinking. Perhaps part of it will be not re-doing work on things that have not changed - like in the Rust compiler library you linked - but part of it will have to be improving the algorithms. 

There was a short related discussion on the forum recently https://solvespace.com/forum.pl?action=viewthread&parent=4967&tt=1673005989

phkahler, 2023-01-07:
@ruevs  I've got 2 old hard drives sitting in front of me. Hoping to recover some old code I wrote with a very efficient spatial index for finding geometry at a point. If it's not too ugly it might help several O(n^2) algorithms in solvespace to become O(n*log(n)). fingers crosses it's all still on there...

ruevs, 2023-01-07:
> some old code I wrote with a very efficient spatial index for finding geometry

Ahh you've mentioned this before - I remember.

My time lately is very limited - sorry for that. I have a huge list of things in SolveSpace I want to work on and never get to it :-(

hinell, 2023-01-13:
@phkahler Sad you didn't backup your repo. Anyway, the least problem SolveSpace has is the spatial geometry calculations.