solvespace/solvespace #1195: Solvespace crashes on Pinephone!

bryanpaget, 2022-01-22:
### System information

#### SolveSpace version

The solvespace packages in postmarketOS are called:

`solvespace-3.0-r0`
`solvespace-dev-3.0-r0`

And `solvespace-cli version` returns `SolveSpace version 3.0~0b7b7b70`.

#### Operating system

`Linux pine64-pinephone 5.15.13 #2-postmarketos-allwinner SMP PREEMPT Sun Jan 16 01:17:53 UTC 20 aarch64 Linux`

### Expected behavior

I'm trying to test solvespace on the Pinephone! I'm just trying the launch the graphical interface to see how it fits on the Pinephone screen.

### Actual behavior

When I try to launch solvespace by tapping on the icon, nothing happens. When I try to launch solvespace from the terminal app I get the following:

```
pine64-pinephone:~$ solvespace
SolveSpace!

Killed
```

[image]

When I watch `htop` I can see the RAM usage go up all the way and then the program crashes. Perhaps it's a memory leak?

### Additional information

This bug post is in response to a Tweet I made:

https://twitter.com/bryan_paget/status/1483986134631915526

Which then led to someone asking me to test out solvespace.

https://twitter.com/SolveSpace/status/1484277626789412865

kk6mrp, 2022-01-22:
That looks like maybe @Symbian9's request. I attempted building Solvespace instead of installing from the Alpine repo (as it appears you have) a while back and it did run except it was an empty window with the menubar only. At which point I lost interest, I'll have to try again and actually try troubleshooting it.

Can you try building from source?

nabijaczleweli, 2022-01-22:
`ulimit -Sc unlimited`, get a dump, trace it in gdb?

kk6mrp, 2022-01-22:
```sh pine64-pinephone:~/dev/solvespace/build/bin$ gdb ~/dev/solvespace/build/bin/solvespace ~/core.4051 
GNU gdb (GDB) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-alpine-linux-musl".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/weston/dev/solvespace/build/bin/solvespace...

warning: Can't open file /memfd:wayland-cursor (deleted) during file-backed mapping note processing
[New LWP 4051]
[New LWP 4062]
[New LWP 4063]
[New LWP 4064]
[New LWP 4065]
Core was generated by `./solvespace'.
#0  0x0000ffffb5c90d10 in memcmp (vl=0xffffb53742f4, vr=0xffffb1e5ff90, n=6) at src/string/memcmp.c:6
6	src/string/memcmp.c: No such file or directory.
[Current thread is 1 (LWP 4051)]
(gdb) bt
#0  0x0000ffffb5c90d10 in memcmp (vl=0xffffb53742f4, vr=0xffffb1e5ff90, n=6) at src/string/memcmp.c:6
#1  0x0000ffffb3f82c40 in  () at /usr/lib/libgio-2.0.so.0
#2  0x0000ffffb3f82c80 in  () at /usr/lib/libgio-2.0.so.0
#3  0x0000ffffb3f82c80 in  () at /usr/lib/libgio-2.0.so.0
#4  0x0000ffffb3f82df0 in  () at /usr/lib/libgio-2.0.so.0
#5  0x0000ffffb3f83444 in  () at /usr/lib/libgio-2.0.so.0
#6  0x0000ffffb3f2e414 in  () at /usr/lib/libgio-2.0.so.0
#7  0x0000ffffb3f2f1e0 in g_resources_get_info () at /usr/lib/libgio-2.0.so.0
#8  0x0000ffffb4eaf880 in  () at /usr/lib/libgtk-3.so.0
#9  0x0000ffffb4ff1d10 in  () at /usr/lib/libgtk-3.so.0
#10 0x0000ffffb4ff5008 in  () at /usr/lib/libgtk-3.so.0
#11 0x0000ffffb4f83864 in  () at /usr/lib/libgtk-3.so.0
#12 0x0000ffffb4157604 in g_cclosure_marshal_VOID__OBJECTv () at /usr/lib/libgobject-2.0.so.0
#13 0x0000ffffb4154700 in  () at /usr/lib/libgobject-2.0.so.0
#14 0x0000ffffb4168f64 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#15 0x0000ffffb4169874 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#16 0x0000ffffb4152c30 in  () at /usr/lib/libgobject-2.0.so.0
#17 0x0000ffffb4154700 in  () at /usr/lib/libgobject-2.0.so.0
#18 0x0000ffffb4168f64 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#19 0x0000ffffb41699d8 in g_signal_emit_by_name () at /usr/lib/libgobject-2.0.so.0
#20 0x0000ffffb4ce5f50 in  () at /usr/lib/libgdk-3.so.0
#21 0x0000ffffb4cb81ec in gdk_display_manager_open_display () at /usr/lib/libgdk-3.so.0
#22 0x0000ffffb4f62e98 in gtk_init_check () at /usr/lib/libgtk-3.so.0
#23 0x0000ffffb4f62ec4 in gtk_init () at /usr/lib/libgtk-3.so.0
#24 0x0000ffffb5962084 in Gtk::Main::init(int*, char***, bool) () at /usr/lib/libgtkmm-3.0.so.1
#25 0x0000ffffb59620f4 in Gtk::Main::Main(int&, char**&, bool) () at /usr/lib/libgtkmm-3.0.so.1
#26 0x0000aaaad9fed360 in SolveSpace::Platform::InitGui[abi:cxx11](int, char**) ()
#27 0x0000aaaad9fc9748 in main ()

(gdb) bt full            
#0  0x0000ffffb5c90d10 in memcmp (vl=0xffffb53742f4, vr=0xffffb1e5ff90, n=6) at src/string/memcmp.c:6
        l = 0xffffb53742f4 "theme/"
        r = 0xffffb1e5ff90 "theme/Adwaita/gtk.css"
#1  0x0000ffffb3f82c40 in  () at /usr/lib/libgio-2.0.so.0
#2  0x0000ffffb3f82c80 in  () at /usr/lib/libgio-2.0.so.0
#3  0x0000ffffb3f82c80 in  () at /usr/lib/libgio-2.0.so.0
#4  0x0000ffffb3f82df0 in  () at /usr/lib/libgio-2.0.so.0
#5  0x0000ffffb3f83444 in  () at /usr/lib/libgio-2.0.so.0
#6  0x0000ffffb3f2e414 in  () at /usr/lib/libgio-2.0.so.0
#7  0x0000ffffb3f2f1e0 in g_resources_get_info () at /usr/lib/libgio-2.0.so.0
#8  0x0000ffffb4eaf880 in  () at /usr/lib/libgtk-3.so.0
[... 54 lines cut ...]
     0.95%  solvespace       [unknown]                      [k] 0xffff800010b5460c
     0.81%  solvespace       ld-musl-aarch64.so.1           [.] memcpy
     0.76%  solvespace       ld-musl-aarch64.so.1           [.] memset
     0.72%  solvespace       solvespace                     [.] _mi_ptr_cookie
     0.66%  solvespace       solvespace                     [.] _mi_arena_alloc_aligned
     0.64%  solvespace       solvespace                     [.] mi_segment_commit_mask
     0.62%  solvespace       libglib-2.0.so.0.7000.1        [.] g_slice_alloc
     0.62%  solvespace       ld-musl-aarch64.so.1           [.] get_meta
     0.61%  solvespace       solvespace                     [.] mi_stat_update
     0.59%  solvespace       ld-musl-aarch64.so.1           [.] strcmp
     0.59%  solvespace       solvespace                     [.] __aarch64_ldadd8_acq_rel
     0.58%  solvespace       solvespace                     [.] mi_segment_reclaim_or_alloc
     0.54%  solvespace       ld-musl-aarch64.so.1           [.] __lock
     0.46%  solvespace       libglib-2.0.so.0.7000.1        [.] g_str_hash
     0.46%  solvespace       solvespace                     [.] mi_os_mem_alloc_aligned
     0.45%  solvespace       [unknown]                      [k] 0xffff800010023724
     0.45%  solvespace       libglib-2.0.so.0.7000.1        [.] g_slice_free_chain_with_offset
     0.44%  solvespace       ld-musl-aarch64.so.1           [.] __unlock
[... cut ...]
     0.40%  solvespace       solvespace                     [.] __aarch64_ldadd8_relax
     0.40%  solvespace       solvespace                     [.] mi_segments_page_alloc
     0.39%  solvespace       libglib-2.0.so.0.7000.1        [.] g_ascii_strncasecmp
     0.38%  solvespace       ld-musl-aarch64.so.1           [.] __libc_malloc_impl
     0.38%  solvespace       ld-musl-aarch64.so.1           [.] alloc_slot
     0.38%  solvespace       solvespace                     [.] mi_segments_track_size```

Thovthe, 2022-04-16:
Have you tried this on Manjaro?  I have made a PKGBUILD that can be found in #1233.  Runs on the PbP with only occasional rendering issues.

Thovthe, 2022-04-17:
I have compiled solvespace for Manjaro on the Pinephone, found a couple more implicit build dependencies but it starts without any apparent memory leaks.  The windows start with all the gtk supplied menus seeming to work but the viewport and property browser have a white field with "Unable to create a GL context" in the center.  Maybe an implicit runtime dependency not on a default install?