solvespace/solvespace #1196: Part level configuration could be the answer to many implementation issues and lead to a common standard

barkYboY, 2022-01-23:
A part level configurations feature would be very useful. With this feature a part(imported geometry) in a model could shift things around without having to open, modify, save, then reload the imported geometry. Since the part is not changing, just the used configuration, the part doesn't need to be modified. This feature not only would increase workflow productivity but could also be used to solve some other implementation issues (seen mentioned in other threads).

  Configurations could help enable parts libraries by being able to adjust a standard model without having to change it. Say screws for example could have different configs for at least lengths, but potentially many other things. A hole wizard implementation could use different screw configs for tap diam, clearance, counter sink etc, to encapsulate the geometry differences in solids... I've used configurations in Solidworks to successfully solve tolerance stack problems. Configurations could be used to hold versions of parts too. So they could evolve and be rolled back to simpler states with less confusion.

  If in addition to configurations a locking (key based encryption) feature was added, Solvespace could potentially get much more adoption from larger organizations, if not become the standard. Enabling the ability to protect and segregate part IP(how the part was designed) from how to implement it, is one of the main hurdles of hurdles for a mainstream 3D CAD standard. Anybody that has tried to design around a broken step file knows what I'm talking about.

  Here's an analogy. Suppose a cad standard is the road and the parts are the vehicles that travel on roads. The majority of those making vehicles will make them comply with standard roads.  More entities make vehicles than make roads because many more vehicles are purchased than roads built.  You'll still have niche special vehicles for special roads (like racing and high end cad systems). Yet these will usually stem from an agreed upon standard and the majority of vehicle makers will adopt compliance with the most common road that enables the largest sales market. If some of the details of a road are hidden from the masses so only certain vehicle makers have unfair advantages, that road will probably remain proprietary and likely cannot become the agreed upon standard no matter how great it is. In my opinion, Solvespace has the opportunity to become the standard road for 3D modeled cad. A road that can lead to an 80% or more increase in efficiency of parts usage for people through on demand manufacturing. Locking and encrypting the internal designs of a part will enable the IP to be protected.  This protection will grant the confidence and security needed for organizations to trust putting their assets on this road.  I believe configurations are an important step towards enabling protection of internal design details while still retaining the ability to share the needed aspects of the design.

  Imagine nearly every item you can buy having a downloadable model for easy integration into your project.  Next visualize distributed micro factories that can produce 70-80% of our parts (electrical, plumbing, appliance components...) as needed so you don't have to ship parts hardly at all(or store them).  History attributes much of the Roman empire's success to that of it's roads and how it could move it's resources around.  Solvespace can be the next key component of our emerging virtual roads that enable most manufactured goods to be produced where they are needed on demand with significantly minimized waste. 

Thank you to all who contribute.

phkahler, 2022-01-24:
This is related to issues #77  #80, and maybe #85.

From an implementation perspective, the part absolutely IS changing. The saved part becomes an incomplete template of sorts - one where we can override parameters for each instance of it. This stuff is definitely on the radar for implementation some day.

WizardUli, 2022-12-05:
Is there some workaround / alternative workflow for this in current SolveSpace?
I'm coming from OpenSCAD where instantiating a part multiple times just with different parameters is one of the most common things ever. (But you can't query generated geometry in any way let alone use something like constraints for assembling parts together which is why I'm looking at SolveSpace right now.)

closed 2025-05-16 by phkahler