solvespace/solvespace #1199: Building on macOS 10.14.6  requires root?

ruevs, 2022-01-24:
### System information

- **SolveSpace version:** git master b399d9a
- **Operating system:** clean install of macOS Mojave 10.14.6, Xcode 11.3.1 (the latest supported on Mojave) in a VirtualBox virtual machine.

### Expected behavior

Building on the command line [according to the instructions](https://github.com/solvespace/solvespace#building-on-macos) should work.

### Actual behavior

First I had to install libomp - so I created a pull request to describe that https://github.com/solvespace/solvespace/pull/1198. This is OK and the [CI scripts already do it](https://github.com/solvespace/solvespace/blob/b399d9a1ec50086cdfc827a43d661ef79052e175/.github/scripts/install-macos.sh#L3) .

Then I had to run the `cmake` and `make` commands as root (sudo)!?

```
sudo cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_OPENMP=ON -DENABLE_LTO=ON
sudo make -j4
```

Otherwise I get errors:
For `cmake`:

@vespakoen any idea?

On the other hand doing
`cmake .. -G Xcode` and building from the Xcode GUI works.

### Additional information

The macOS virtual machine was created using [this script](https://github.com/myspaghetti/macos-virtualbox).
I'm not an expert neither on Mac nor on Linux so I may be doing something stupid.

vespakoen, 2022-01-27:
What are the permissions on the libomp.dylib file?

Mine looks like this:

```
➜  solvespace git:(master) ✗ ls -lah /opt/homebrew/Cellar/libomp/13.0.0/lib/
total 2904
drwxr-xr-x  4 koen  admin   128B Jan 25 23:10 .
drwxr-xr-x  8 koen  admin   256B Nov 23 00:15 ..
-r--r--r--  1 koen  admin   816K Sep 24 18:18 libomp.a
-r--r--r--  1 koen  admin   630K Jan 25 23:10 libomp.dylib
```

I can run cmake without root fine over here at least (and in CI as well).
Not sure what's going on...

ruevs, 2022-02-06:
```
as-MacBook-Pro:~ a$ cd /usr/local/include/../lib/
as-MacBook-Pro:lib a$ ls -la libo*.*
lrwxr-xr-x  1 a  admin  36 Jan 24 17:58 libomp.a -> ../Cellar/libomp/13.0.0/lib/libomp.a
lrwxr-xr-x  1 a  admin  40 Jan 24 17:58 libomp.dylib -> ../Cellar/libomp/13.0.0/lib/libomp.dylib
as-MacBook-Pro:lib a$ cd ../Cellar/libomp/13.0.0/lib/
as-MacBook-Pro:lib a$ ls -la
total 2856
drwxr-xr-x  4 a  admin     128 Sep 24 19:18 .
drwxr-xr-x  8 a  admin     256 Jan 24 17:58 ..
-r--r--r--  1 a  admin  841744 Sep 24 19:18 libomp.a
-r--r--r--  1 a  admin  616132 Jan 24 20:58 libomp.dylib
```

ruevs, 2022-02-06:
```
as-MacBook-Pro:lib a$ cd /Users/a/solvespace/git/build/extlib/zlib/CMakeFiles/zlibstatic.dir/
as-MacBook-Pro:zlibstatic.dir a$ ls -la
total 1536
drwxr-xr-x  43 root  staff    1376 Jan 24 21:00 .
drwxr-xr-x   8 a     staff     256 Jan 24 20:58 ..
-rw-r--r--   1 root  staff    2772 Jan 24 18:41 DependInfo.cmake
-rw-r--r--   1 root  staff    2484 Jan 24 18:41 adler32.o
-rw-r--r--   1 root  staff   12039 Jan 24 18:41 adler32.o.d
-rw-r--r--   1 root  staff   29605 Jan 24 18:41 build.make
-rw-r--r--   1 root  staff    1437 Jan 24 20:58 cmake_clean.cmake
-rw-r--r--   1 root  staff      43 Jan 24 20:58 cmake_clean_target.cmake
-rw-r--r--   1 root  staff  176167 Jan 24 21:00 compiler_depend.internal
-rw-r--r--   1 root  staff  193716 Jan 24 21:00 compiler_depend.make
```

vespakoen, 2022-02-06:
Interesting... not sure what is going on though, I'll suggest running / looking up the:

brew doctor

and

brew prune

commands, and give those a try.. also make sure to remove the build folder before rebuilding (using sudo)

ruevs, 2022-02-06:
> also make sure to remove the build folder before rebuilding (using sudo)

Great idea @vespakoen! That fixed it! 
Doing:

```
sudo rm -rf build
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_OPENMP=ON -DENABLE_LTO=ON
make -j4
```

worked just fine! But why?! Eh well...

Anyway just FYI:

> Interesting... not sure what is going on though, I'll suggest running / looking up the:
> 
> brew doctor
> 
> and
> 
> brew prune
> 
> commands, and give those a try.. 

```
as-MacBook-Pro:~ a$ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: You are using macOS 10.14.
We (and Apple) do not provide support for this old version.
You will encounter build failures with some formulae.
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels. You are responsible for resolving
any issues you experience while you are running this
old version.
```

```
as-MacBook-Pro:~ a$ brew prune
Error: Unknown command: prune
```

```
as-MacBook-Pro:~ a$ brew --version
Homebrew 3.3.12
Homebrew/homebrew-core (git revision 2fbf6efd745; last commit 2022-01-24)
```

closed 2022-02-06 by ruevs