solvespace/solvespace #1200: [Wiki] Platforms support & Packaging status

Symbian9, 2022-01-27:
### System information

- **SolveSpace version:** all
- **Operating system:** various

### Expected behavior

List *all possible platforms* which may be used to run *SolveSpace*.

> *NOTE:* This issue is not a feature request, but just documentation thread for collect all info related to title.

### Actual behavior

Actually *SolveSpace* target platforms are Linux, macOS & Windows.

But there are already a lot of other platforms which may be (and already) used to run *SolveSpace*

### Additional information

| Platform | Compiling instructions | Packaging Status | Notes |
| :--- | :--- | :---: | :--- |
| Raspberry Pi | 1. https://solvespace.com/forum.pl?action=viewthread&parent=3955 <br> 2. https://solvespace.com/forum.pl?action=viewthread&parent=3796 | TBA | TBA |

ruevs, 2022-01-27:
So:
- Raspberry Pi is Linux and  3.0 rc2 and 2.3 are already in the official package manager (http://archive.raspbian.org/raspbian/pool/main/s/solvespace/).
Compiling following the Linux instructions should just work - [it does not, but I'll make a pull request that will make it "just work™"](https://solvespace.com/forum.pl?action=viewthread&parent=3955).
-  FreeBSD - SolveSpace 3.0 is already packaged (https://www.freshports.org/cad/solvespace/).
*BSDs are Unixes and the [OpenBSD instructions](https://github.com/solvespace/solvespace#building-on-openbsd) or the [Linux instructions](https://github.com/solvespace/solvespace#building-for-linux) should work/help with compiling. That said BSD users usually are qualified enough to manage it.
- Haiku - OK when someone ports it to Haiku they will presumably submit a pull request.

So what it the point of this issue?

Symbian9, 2022-01-27:
> So what it the point of this issue?

As I noted:

> NOTE: This issue is not a feature request, but just `documentation` thread created to collect all info related to the title.

Symbian9, 2022-01-27:
**UPD:** Inkscape ported to Haiku!

- https://discuss.haiku-os.org/t/inkscape-now-available-in-package-repositories/11850

So, it seems like GTK3 porting is complete and SolveSpace porting is finally possible:

- https://discuss.haiku-os.org/t/xlibe-an-xlib-x11-compatibility-layer-for-haiku/11692/113
> And with that, my X11 compatibility / GTK3 porting adventure is complete. I’ll let the HaikuPorts team take it from here, and work on getting more GTK3 applications (GIMP, etc.) available.

ruevs, 2022-01-27:
> > Haiku - OK when someone ports it to Haiku they will presumably submit a pull request.
> 
> Of course, no. Haiku developers packaging team not does creates pull request to original projects repos. Why they should do it "presumably"?

  If the port requires changes to the build system or the core that can be merged upstream they would (I presume) submit a pull request. If not - great - nothing to do.

> 
> > So what it the point of this issue?
> 
> As I noted:
> 
> > NOTE: This issue is not a feature request, but just `documentation` thread created to collect all info related to the title.

OK, but in fact it does not contain useful "info" for end users - any user should be able to look in the package manager / installer / app store / whatever of their OS and see if SolveSpace is available.

That said - you are of course welcome to try to collect up to date packaging status for different operating systems here - it could be useful for someone but [there are better ways for Linux distributions](https://distrowatch.com/dwres.php?resource=compare-packages&firstlist=ubuntu&secondlist=debian&firstversions=0&secondversions=0&showall=yes#allpackages) and Google is always there ;-)

closed 2022-01-27 by ruevs

ruevs, 2022-01-27:
The Repology link is useful and I will add it to the pinned issue.

Symbian9, 2022-01-27:
>  it could be useful for someone but there are better ways for Linux distributions and Google is always there ;-)

@phkahler, May you enable GitHub Wiki for actual SolveSpace repo?

For example, like this:

- https://github.com/Symbian9/azpainter/wiki/Packaging-status

SolveSpace Wiki is really missed as part of docs.