solvespace/solvespace #1202: Time for version 3.1RC1 soon?

phkahler, 2022-01-30:
Asking  @ppd  @rpavlik @ruevs @vespakoen  are there any things you'd like to get in prior to a version 3.1?  We should be able to get it out less than a year after 3.0, which would be a nice change of pace.

For me, I've got one bugfix PR pending that I want in, but I'm not going to get mirror groups is this time much to my disappointment. I think there is enough good stuff for a 3.1 though. I'm curious about the MacOS touchpad issue and will try Edge on old MBA soon. Also would like to see the release as distro friendly as possible, so input specifically from @rpavlik  and @ppd on that would be good.

Thanks,
Paul

vespakoen, 2022-01-30:
I will mess around with the trackpad issue in the coming evenings. I think a setting / checkbox can solve the current issue and provide very nice panning and zooming on mac.
I think it makes sense to include it in the release because the current behaviour on edge is confusing.

ruevs, 2022-01-31:
I could reproduce this https://github.com/solvespace/solvespace/issues/681 by deleting everything from the registry. And would like to debug it.

This one is rather annoying as well https://github.com/solvespace/solvespace/issues/1143

Symbian9, 2022-01-31:
> are there any things you'd like to get in prior to a version 3.1? We should be able to get it out less than a year after 3.0

Should we call translators? (so may update locales just before `3.1` release)

rpavlik, 2022-02-01:
Yeah probably good to run the update translations job and make call for translation. I'll look and make sure all my Debian patches have made it in.

Symbian9, 2022-02-01:
> Yeah probably good to run the update translations job

Anybody may update it today in `master`?

ruevs, 2022-02-01:
> Yeah probably good to run the update translations job and make call for translation.

Done here https://github.com/solvespace/solvespace/commit/a1b2db5e187eb85b85e7ac407e77935973893809

ppd, 2022-02-03:
> * let's make sure to upload a full tarball and not just use the github one, because that excludes submodules and is thus less useful.

I'm thinking this should be part of CI for tagged commits. I can give it a stab sometime soon.

phkahler, 2022-02-03:
> * Please name the release `3.1~rc1`  - if you use a `.`, it gets ordered _after_ 3.1 rather than before.

That's a really useful tip.

I think our mimalloc is currently at 2.0.2 and maybe should be bumped to v2.0.3 which they have tagged.

phkahler, 2022-05-01:
@rpavlik and @ppd  This is the "instructions" I previously followed to udpate mimalloc:
https://github.com/solvespace/solvespace/pull/661#issuecomment-664001203

If there isn't a better way I'll do that again to get us up to their latest tagged version (2.0.7 or so).

phkahler, 2022-05-12:
@rpavlik  it looks like the version in CMakes.txt or whatever is displayed in the about dialog. It is currently showing 3.1 even thought there is no tag with that yet.

Github refuses to let me create a 3.1~rc1 or v3.1~rc1 tag.  It suggests v3.1-alpha so it seems the ~ is the problem can I use the - character instead?

rpavlik, 2022-05-12:
Oh right, it always converts them to underscores. Either is fine as long as I do the debian import

phkahler, 2022-05-12:
Just tagged v3.1-rc1 and it looks like @ppd changes worked nicely. I haven't opened the tarball yet but it's big so maybe the dependency sources made it! We should be at mimalloc 2.0.6 as well.

Thanks everybody. Let me know something is actually broken that I didn't notice ;-)

rpavlik, 2022-05-12:
Looks good to me so far, got it imported into the debian git just fine, package almost done.

phkahler, 2022-05-26:
@rpavlik   If you're happy with it from a packaging side I think I'll drop the rc this weekend and tag 3.1. Are we good?
I'm looking forward to this one getting into more distros over the next year ;-)

BTW We made the front page of HN a few days ago and 3.1-rc1 has a bunch of downloads:

ruevs, 2022-05-26:
I'm fine with releasing 3.1 the issues I mentioned above https://github.com/solvespace/solvespace/issues/1202#issuecomment-1025700337 still bug me but I obviously did not find time to work on them for almost 5 months :-(

rpavlik, 2022-05-26:
oh, did we want to rename the default branch to "main" as the tendency seems to be? and are all the translations updated? I didn't run the gettext thing this time.

ruevs, 2022-05-26:
> and are all the translations updated? I didn't run the gettext thing this time.

I ran it 4/months ago and since then only Turkish and German .po were touched. I'll try to do it gain tonight. https://github.com/solvespace/solvespace/commits/master/res/locales

On "main" vs. "master" - the whole "political correctness" "movement" is a bit much to me, but ultimately I am ambivalent to it.

ruevs, 2022-05-27:
I ran the translations update - nothing changed, but the dates in the .po files and "Project-Id-Versionan" to 3.1 in `res/messages.pot`. So it's fine.

phkahler, 2022-06-01:
3.1 is still running through CD, but will come out fine. @ruevs if you want to update the main solvespace page to include version 3.1 in the table I won't complain ;-) Otherwise I'll get to it hopefully tonight.

closed 2022-06-01 by phkahler

ppd, 2022-06-01:
Huh, seems that the tag wasn't published when the Snap CD job ran. Because of that, the build got marked as unstable, which prevents us from pushing it into the stable channel. Hm. Strange. That used to work, I think

ppd, 2022-06-01:
@phkahler I already did, but locally. Nothing to worry about for now. But for the next release, we need something better.

ruevs, 2022-06-02:
> @ruevs if you want to update the main solvespace page to include version 3.1 in the table I won't complain ;-)

Done here https://github.com/solvespace/solvespace-web/pull/35. @jwesthues should publish to the web site after it is merged.

jwesthues, 2022-06-03:
Very cool, just pulled to live on the website. Let me know if anyone sees any issues.