solvespace/solvespace #1205: NURBS hexagonal segment union solid fails

Symbian9, 2022-01-31:
### System information

- **SolveSpace version:** `3.0~`https://github.com/solvespace/solvespace/commit/b399d9a1ec50086cdfc827a43d661ef79052e175
- **Operating system:** Debian 10 buster

### Expected behavior

There should not be fails in NURBS segment union solid

### Actual behavior

NURBS solid fails.

### Additional information

- [slvs3x-hex-hole_asm.zip](https://github.com/solvespace/solvespace/files/7974122/slvs3x-hex-hole_asm.zip)

<details><summary>Sreenshot:</summary>

[image]

</details>

NOTE: Instead of https://github.com/solvespace/solvespace/issues/1204 (where same metric segment solid used), here I used corrected cut solid as a segment for hexagonal hole.

simrim1, 2022-02-01:
Click the group g004-rotate to show its properties. Then check the property "force NURBS surfaces to triangle mesh". It will fix the error.

[image]
.

Symbian9, 2022-02-01:
> Then check the property "force NURBS surfaces to triangle mesh". It will fix the error.

You forced triangle mesh. That is out of scop of this *NURBS solid bug*.

simrim1, 2022-02-01:
@Symbian9, That is correct. I am new here and sorry if it was out of scope of your issue. I am deleting my reply. Sorry again.

phkahler, 2022-02-02:
@simrim1 dont worry about it :-)  he is correct though. When NURBS surfaces fail, one "solution" is to force to triangle mesh but that is only viable in some cases. You won't be able to export a STEP file for example. Even though it masks a NURBS issue, there is still an issue that we would like to fix.