solvespace/solvespace #1207: Feature request: Change workplane (sketch-in-plane) properties in the Property Browser window...

simrim1, 2022-02-01:
Hi,

This is a feature request for changing the workplane (sketch-in-plane) properties in the Property Browser window.

On selection of sketch-in-plane group name it will show properties as follows

First property will be Move with three text fields (X, Y, Z) to move the Origin Datum Point from its current position according to the the values in X, Y, Z text fields. The values will obviously be Numbers Plus or Minus.

The second property will be Tilt, again, with three text fields (X, Y, Z) which will tilt workplane from its current position in degrees around X, Y, Z axis according to the values entered in the text fields.

The third property will be Rotate the workplane with two text fields Clockwise or Anti-Clockwise in degrees.

This will help in setting the workplane at any poition/orientation without a lot of hassle and without using any construction/geomatery lines.

Thanks

phkahler, 2022-02-03:
I'm not in favor of this. It presumes a particular way to define the position and orientation of a workplane. In fact your wording suggests that the (X,Y,Z) values are relative to the existing location, so they would make zero sense after the workplane moves. Absolute coordinates seem like a slightly better idea. What about adding a constraint for points to set their X,Y,Z position and display it on the sketch? That has come up before. As for orientation, I'm not sure what's best. Most people constrain the orientation relative to something else on the screen, so the current method of constructing it seems pretty good.

simrim1, 2022-02-03:
Thanks for your response and suggestion about datum point. I believe it will also do. But the other two options make a lot of sense. The text fields suggested above will be decorated with up/down arrows where mouse wheel can be used for increasing/decreasing numeric values. As the values are changing in the text field, the workplane will tilt, rotate so the user can see where it is as per values.

I know there are ways to set the orientation of workplane but it hinders productivity. Construction lines make sense in sketch itself but just to orient a workplane you keep adding construction lines and their constraints. It does the job but it makes the drawing a mess with construction lines here and there.  Whereas you can create a workplane on any of the main coordinate planes (XY, YZ, XZ) and then tilt rotate it according to your needs if need be. And Yes, it can be bound to any of the main coordinate planes and it will preserve the values of Tilt, Rotate with respect to the main coordinate plane. So when you click it you will see the values in properties.

In some cases it becomes a necessity. Lets suppose you make a sketch and extrude it. Then you rotate the view and you set the view with approximation with mouse. Then you select workplane from Sketch->Workplane.  Then you press Shift+W by selecting only one point as origin to sketch in the workplane. You make your sketch and keep going with your drawing. And then you want to tilt, rotate the workplane. You can't do it now. You will have to delete the workplane which will eventually delete the sketch. If you had tilt, rotate values in the workplane you would simply change them and adjust the workplane's orientation.

Does it make sense what I said?

Thanks

phkahler, 2022-02-04:
> Lets suppose you make a sketch and extrude it. Then you rotate the view and you set the view with approximation with mouse. Then you select workplane from Sketch->Workplane. Then you press Shift+W by selecting only one point as origin to sketch in the workplane.

This was new to me, I didn't realize this was possible. But I do see some issues:
1) The workplane (request) created in the extrude group is nearly useless. It does have a point and normal shown in the text window if you select it, but there is very little that can be done to set its alignment aside from dragging the normal vector on the diagram. You can constrain that to some geometry in the existing sketch though using point-on-plane constraints.
2) After a new group (sketch in new workplane) is created by selecting the workplane in the extrude group, going back and changing the normal of the workplane (request) does not seem to update the normal of the workplane (group).

I would say that the usefulness of a workplane entity should be improved, and that issue 2 here is a bug.

What you seem to want is to draw in a workplane and then use the elements drawn in the plane to align the workplane. Is that correct? If so, that is not possible because the workplane orientation is defined in a previous group, and constraints can not modify entities from previous group in order to satisfy them.

simrim1, 2022-02-05:
I think I confused it. Sorry about that. I actually wanted to refer to "sketch-in-plane" rather than workpalne. sketch-in-plane is defined on a workplane. So, the properties I mentioned above will apply to sketch-in-plane and not workplane.

Please note, I am not saying that what I want to achieve is impossible with current arrangements.  It is achievable with construction lines to define a sketch-in-plane.

Actually a sketch-in-plane can be defined in two ways. 

1. Select Origin and press Shift+W and it will automatically align the sketch-in-plane with one of the three predefined workplanes (XY, YZ. XZ) depending on which plane is nearer to the viewer/user. Here you have no control over the orientation of the new sketch-in-plane.

2. Select two lines and a point and press Shift+W and it will align a new sketch-in-plane with the two selected lines. These lines can be design and/or construction lines or combination of both. This way it is your responsibility to draw the lines (if need be) and set their angles so your new sketch-in-plane is oriented correctly. Here you have some control (but not full) over the orientation of the new sketch-in-plane.

What I am suggesting is that we should have full control over the orientation of new sketch-in-plane. Now the question is how we can achieve this.  The answer is in my first post i.e. with the help of text fields in the properties of sketch-in-plane group.

Now another question is will we achieve something extraordinary with this suggested arrangement? The answer is yes. Please pay attention and I will try to explain it as much as I can. But if there is still some confusion I can elaborate it on some sort of online meeting with some arrangement of screen sharing.

Let's suppose we create a cube (solid). now on every edge of the cube we want to make holes at different angles. We will have to create construction lines on four corners to set the required angles. This will be extra work on each corner to create sketch-in-plane and draw circles for holes. 

If we had sketch-in-plane group properties we could just create a sketch-in-plane using the first method i.e. selecting the origin point only and pressing Shift+W. It will create a new sketch-in-plane on the workplane nearest to the viewer/user as it does now. Then we can change the orientation of the sketch-in-plane group properties to change tilt angle on any axis and rotate it clock-wise and anti-clockwise if required.

No need to draw construction lines and the method will be clean and smooth. Now a big question is, the tilt and rotate angles will be relative to what? It will be relative to the workplane automatically selected at the time of selecting origin point and pressing Shift+W and it will always be bound to that workplane. So whenever we see the properties of sketch-in-plane it will retain the angles of tilt and rotation. On the top it will be showing that selected sketch-in-plane is bound to XY, YZ or XZ workplane. And the angles of tilt and rotate showing there in text boxes will be relative to the workplane this sketch-in-plane is bound to. And we can change these angles anytime. And the design will be updated automatically.

I hope I have explained it properly and if there is still some confusion please let me know I will try again to clear it.

Thanks