solvespace/solvespace #1210: Change function names (quaternions)

phkahler, 2022-02-05:
This would be strictly a change of naming convention in the code.

Quaternions currently have 3 functions called RotationU, RotationV, and RotationN.

What would you expect they do based on the names? What if I then told you the return a vector? Now what you think they do? There are not comments in the code documenting these (adding some might also be a solution)

A quaternion can be used to apply a rotation to a vector or another quaternion. They can also be used as a compact representation of a coordinate systems orientation. As such, these 3 functions can be used to extract 3 basis vectors from a quaternion. RotationU returns the U axis (I think x) as is the vector (1,0,0) were rotated by the quaternion. Similar for V and N (y and z). With this understanding, the naming makes more sense. The problem is that the naming is not explanitory.

I'm wanting to use BasisU in place of RotationU, but I'm not sure what others think about that. Other suggestions are welcome too. I just dont like the current situation.

Something similar is done with "normals" which are stored as quaternions, and have similar functions with different names that just call the quaternion functions.

Thoughts on this?

Symbian9, 2022-02-05:
> A quaternion can be used to apply a rotation to a vector or another quaternion.

Some related reference info:

1. Quaternions — https://liorsinai.github.io/mathematics/2021/11/05/quaternion-1-intro.html
1.1 https://news.ycombinator.com/item?id=29510237
2. Rotations with Quaternions - https://imadr.github.io/rotations-with-quaternions/
2.1. https://news.ycombinator.com/item?id=27353263
3. Quaternions: The Strange Numbers That Birthed Modern Algebra - https://www.quantamagazine.org/the-strange-numbers-that-birthed-modern-algebra-20180906/
3.1. https://news.ycombinator.com/item?id=23966755
4. Understanding Quaternions - https://www.3dgep.com/understanding-quaternions/
4.1. https://news.ycombinator.com/item?id=18265355
5. How Quaternions encode rotations – derivation and sample code - http://loopspace.mathforge.org/HowDidIDoThat/Codea/Quaternions/
5.1. https://news.ycombinator.com/item?id=12441656
6. Let's remove Quaternions from every 3D Engine - https://marctenbosch.com/quaternions/
6.1 https://news.ycombinator.com/item?id=29512302

hinell, 2022-02-06:
Refactoring.

phkahler, 2022-02-06:
I'm leaning toward VectorU()  VectorV() and VectorN() 

This is with an eye toward something which might be controversial. Changing the members of EntityBase from normal and numNormal to basis and numBasis. This way we get entity.numBasis.VectorU() for example rather than entity.numNormal.RotationU(). But I would definitely put that in a different PR than renaming those "rotation" functions. Another issue besides the extensive name changes is that currently a face normal vector is in face stored in the "normal" quaternion by only using 3 elements of it, in that case the name "normal" makes more sense.

Evil-Spirit, 2022-09-29:
> GetU()? UAxis() or VectorU()?

XYZ ...  up, right, forward ... UVN... Whatever you call it, it will be not obvious anyway. This is transformation basis directions. It is better not to reaname anything and just take some shader math library and use normal math like vec3, mat4, quat and HOLY operator overloading!

ruevs, 2022-10-03:
I agree with @Evil-Spirit that renaming is pointless. Taking "some shader math library" is also pointless in my opinion. The code works - leave it.

What I do is read about quaternions every time (current "optimal" known reference https://eater.net/quaternions  https://www.youtube.com/watch?v=zjMuIxRvygQ) understand them, completely fail to build an "instinctive understanding", use them, forget them. Until next time.

Evil-Spirit, 2022-10-03:
> Taking "some shader math library" is also pointless in my opinion. The code works - leave it.

Function already named properly. Just leave it! If you want to touch - just take normal math library with standartized names and  most of developers will understand code, and will be happy with OUTSTANDING OPERATOR OVERLOADING.

ruevs, 2022-10-03:
> You don't need to understand them, just use them

Of course. But I LIKE to understand them :-) Just recently I ["stole"/reused most of the liner algebra from SolveSpace](https://github.com/ruevs/jartsy/blob/master/src/jartsyla.h) and I did go through it all line by line - just for fun :-)

Evil-Spirit, 2022-10-03:
> through it all line by line - just for fun :-)

Definitely you was needed this. I do this many times, each time I start new project - I started to do vectors from the scratch, trying to improve. You can see at one of my[ attempts here ](https://github.com/Evil-Spirit/Nutmeg/blob/master/CoreSrc/Math/MathCore.h)