solvespace/solvespace #1211: Display circle length in the browser

hinell, 2022-02-07:
[image]

I request to display a circle length in the browser window when selected.

### System information
- **SolveSpace version:** 3.0.rc2+repack1-3
- **Operating system:** Linux/Microsoft

phkahler, 2022-02-07:
Should we also add arc length?

hinell, 2022-02-07:
@phkahler May be useful, I'm not sure. I was making  a gear and found myself calculating its length it manually.
@EkaterinaEvdokimova Екатерина, ты из России?

EkaterinaEvdokimova, 2022-02-07:
@hinell Да, я из России

phkahler, 2022-02-07:
@hinell   I just checked, and you can measure lengths with Analyze->Perimeter in the menu. You select any number of lines, arc, circles (but no points, not sure why those aren't auto-excluded) and it will give the total length.

It still seems reasonable to display the length of a single selected entity in the text window though.

phkahler, 2022-02-07:
> @hinell Да, я из России

I've looked at google search data, and some other things and it seems Solvespace is quite popular in Russia and parts of eastern Europe. Does anyone know why that might be? Just curious.

hinell, 2022-02-07:
@phkahler It's handy to have this information listed right away in the browser data in unit-less format. I used this information to calculate the pitch for the gear (in non standard way of course). Would be best if we can have _cog tool_ for that at some point in the future though.

Regarding geographical distribution of users, well I'm not sure why. Probably because it's free. Personally I started to use it only recently. It's a good (not perfect though) alternative for Linux.

ruevs, 2022-02-08:
There are better ways.

You can type formulas in any dimension field. If you want a circle with a circumference of 500 just type `500/(2*pi)` for the radius.

You can use `sqrt`, `square`, `sin`, `cos`, `asin`, `acos`, `pi`, `+`, `-`, `*`, `/`, `()`

[image]
[image]

And don't ask for the formulas to be remembered it is the oldest and best feature request  https://github.com/solvespace/solvespace/issues/77 ;-)

ruevs, 2022-02-08:
> > @hinell Да, я из России

А аз съм от България :-)

> 
> I've looked at google search data, and some other things and it seems Solvespace is quite popular in Russia and parts of eastern Europe. Does anyone know why that might be? Just curious.

@phkahler SolveSpace simply resonates with the proverbial Slavic soul :-D

Joke aside, my theory is that in the former eastern block there (still) are a disproportionately high percentage of geeks/DIYorselfers/makers/whatever and the minimalist and clean "ideology" of SolveSpace attracts people who do not want to pirate SolidWorks (or any of the others) or their systems simply can not meet the [system requirements](https://www.solidworks.com/support/system-requirements).

hinell, 2022-02-08:
@ruevs [#issuecomment-1032485216](#issuecomment-1032485216)
Omg why in hell isn't this documented in the [main reference](https://solvespace.com/ref.pl)?! :smile: 

>And don't ask for the formulas to be remembered 

Omg sounds like you need a bit of fresh blood to refresh some rusty  SolveSpace gears out here!

Btw, this reminds me of the TypeScript issue that they can't solve for about 7 years now. It's constantly poping up but nobody cares. The issue is simply though....

ruevs, 2022-02-08:
> @ruevs [#issuecomment-1032485216](#issuecomment-1032485216)
> Omg why in hell isn't this documented in the [main reference](https://solvespace.com/ref.pl)?! 😄 In which units `cos` / `sin` / `tan` functions accept value? Radians? It seems like they are working incorrectly.

It is documented a bit (exactly in the reference) ;-)

> ### Dimension Entry and Units
> All dimensions are entered in the current display units. In most places where a dimension is expected, it's possible to enter an arithmetic expression ("4*20 + 7") instead of a single number. 

But adding the supported operations is a good idea.

The trigonometric function parameters are in degrees.

phkahler, 2022-02-08:
> Omg why in hell isn't this documented...

@hinell   And to really blow your mind, the geometric/algebraic constraint solver can use these functions in constraint equations. We don't currently support user input of constraints, only evaluation of expressions.

Symbian9, 2022-02-08:
> Should we also add arc length?

We already has arc length in "Property Browser", but adding circle length would be good too ().

[image]

hinell, 2022-02-09:
@phkahler Not documented too, right? How can be this done?

ruevs, 2022-02-09:
Done

commit 2383a396 (2022-02-09): UI: Display the circumference of a circle in the property browser

Closes #1211
closed 2022-02-09 by ruevs