solvespace/solvespace #1214: Performance drop with a step and repeat rotating

hinell, 2022-02-08:
```
Vector::WithMagnitude(1) of zero vector! 
Generate::REGEN (for bounding box) took 15720 ms
Generate::REGEN took 31897 ms
Generate::ALL (for bounding box) took 16577 ms
Generate::ALL took 33203 ms
Generate::DIRTY (for bounding box) took 15852 ms
Generate::DIRTY took 32006 ms
Generate::ALL (for bounding box) took 16113 ms
Generate::ALL took 32466 ms
```

See linked model for reproduction. The log is above. The culprit is the `step and repeat rotating` group. If I apply it to any complex sketch (say 20 construction + 10 actual lines) it slows down significantly so I have to wait a few minutes for response. 

**Note:** The `Vector::WithMagnitude(1) of zero vector!` string above is repeated after every line many times. I omitted it for readability.

## Image & file
* [gear-32-2.zip]
[image]

[gear-32-2.zip]: https://github.com/solvespace/solvespace/files/8019594/gear-32-2.zip
### System information
- **Operating system:** Linux Kubuntu
- **SolveSpace version:** 3.0.rc2+repack1-3
### Expected behavior
* Lighting fast overall performance

### Actual behavior
* Very slow step and repeat rotating. It also slows down:
* * solid suppression (browser)
* * toggling control points visibility (browser) 
* * workplane switch (browser)
* * creation and delete of a sketch in a new workplane (editor window and browser)

### Additional information

ruevs, 2022-02-08:
https://github.com/solvespace/solvespace/issues/1186#issuecomment-1032277333
https://github.com/solvespace/solvespace/issues/1186#issuecomment-1032472416

ruevs, 2022-02-08:
@hinell try an edge build.
https://github.com/solvespace/solvespace#via-automated-edge-builds

Since you are on Linux try the edge build from the SnapStore https://github.com/solvespace/solvespace#via-snap-store

Is it still slow?

Can we use your model in a potential future benchmark? Under what license?

ruevs, 2022-02-08:
On current [master](https://github.com/solvespace/solvespace/commit/d8b5281f) changing the repeat from 1 to **_64_** in `g00b-rotate` takes `Generate::DIRTY took 2871 ms` ;-)

[gear-64-2.zip](https://github.com/solvespace/solvespace/files/8022949/gear-64-2.zip)

[image]

By the way there are better ways to draw the teeth of the gear ;-)
https://solvespace.com/forum.pl?action=viewthread&parent=3911&tt=1633767543
https://solvespace.com/forum.pl?action=viewthread&parent=2477&tt=1541434762
https://solvespace.com/forum.pl?action=viewthread&parent=916&tt=1439146696
https://solvespace.com/forum.pl?action=viewthread&parent=291&tt=1413300884
https://github.com/solvespace/solvespace/issues/550#issuecomment-655814031
etc.

hinell, 2022-02-08:
@ruevs Thanks, I will try to do that.

Thanks for the links as well. Turns out, there is a nice article in the [wikipedia on that](https://en.wikipedia.org/wiki/Involute_gear). Cool shit.

Symbian9, 2022-02-08:
> On current `master` changing the repeat from `1` to `64` in `g00b-rotate` takes `Generate::DIRTY took 2871 ms`

@ruevs, as for me, ***NURBS solid difference fails*** in your  SLVS-file from [gear-64-2.zip](https://github.com/solvespace/solvespace/files/8022949/gear-64-2.zip)

[image]

Symbian9, 2022-02-08:
> Turns out, there is a nice article in the [wikipedia on that](https://en.wikipedia.org/wiki/Involute_gear). Cool shit.

JFTR, Here is collection of tutorials for making "Involute Gear" in SolveSpace:

- https://homehack.nl/solvespace-involute-gear/

ruevs, 2022-02-08:
The performance with "Hide edges of solid model" is practically the same `Generate::DIRTY took 2723 ms` - this model is solver limited. Not OpenGL, triangulation etc...

As for the NURBS problem I do see the red edges as well, but after pressing ESC the solid is water tight/valid and we have much simpler models reproducing this. Thus it is not interesting.

hinell, 2022-02-08:
@Symbian9 

> Can we use your model in a potential future benchmark? Under what license?

Sure. As long as you are not trying make profits out it. Thanks for checking it out anyway. ;)

Looks like I'm going to build solvespace from scratch because I also have discovered a few nutty cases that simply hang entire app. No memory leaks however. Surprisingly.

ruevs, 2022-02-08:
> a few nutty cases that simply hang entire app.

If you find reproducible "nutty cases" open issues.

>  No memory leaks however. Surprisingly.

Quite a bit of work has gone into it over the years. You'll find that "edge" is usually as stable as any release.

Symbian9, 2022-02-08:
> `@Symbian9`
> > Can we use your model in a potential future benchmark? Under what license?
> Sure. As long as you are not trying make profits out it. Thanks for checking it out anyway. ;)

Guess, it was reply to @ruevs question ;)

ruevs, 2022-02-10:
So @hinell did you try? Can I we close this now?

hinell, 2022-02-10:
@ruevs Give me a bit more time.

closed 2022-02-10 by ruevs

hinell, 2022-02-10:
@ruevs Yeah, feel free to close this. 
I can confirm that with newer version it became much more responsive. At least it doesn't freeze to an unusable degree for now.

hinell, 2022-02-10:
@ruevs Yeah, feel free to close this. 
I can confirm that with a newer version it became much more responsive.
At least it doesn't freeze beyond unusable state for now.