solvespace/solvespace #1217: "Black" color always shown flat

Symbian9, 2022-02-09:
### System information

- **SolveSpace version:** `3.0~`https://github.com/solvespace/solvespace/commit/a1b2db5e187eb85b85e7ac407e77935973893809 ; `4.0`
- **Operating system:** Debian 10 buster; all supported

### Expected behavior

Redesign color implementation and diffuse shading in rendering, so even solids with "Black" color (`0, 0, 0`) should not look *'flat'* if "ambient lightning" < `1.0`.

### Actual behavior

Actually "Black" color (`0, 0, 0`) always looks fully *'flat'*, even "ambient lightning" = `0.1`:

- Solid color: `1, 1, 1` ("White") / Background color: `0, 0, 0` ("Black")
- Solid color: `0, 0, 0` ("Black") / Background color: `1, 1, 1` ("White")

### Additional information

REFERENCE:

- https://en.wikipedia.org/wiki/Shader
- https://thevirtualinstructor.com/shading-techniques-basics.html
- https://www.creativeartcourses.org/shading-3d-objects/
- https://docs.blender.org/manual/en/2.79/render/blender_render/materials/properties/diffuse_shaders.html
- http://what-when-how.com/3d-animation-using-maya/shading-material-types-refinements-3d-animation-using-maya/

phkahler, 2022-02-09:
All shading in Solvespace is done using nothing more than Snell's law (diffuse) with 2 lights and ambient. Black will just be black because the lighting model is so simple. I did some experiments on one of my branches to implement a more realistic lighting model including the Fresnel term which tends to highlight silhouette edges better and it looked a little better. It's hard to validate that the math is right so I've been tagging (so I can remember) projects which include a PBR based lighting model and GPLv3 compatible license (please don't offer suggestions). My internal list of stuff includes adding one of these better material/lighting models. I also want to add the ability to overlay a heatmap type coloring for FEA results. So I figure all of that can happen with one deep dive into the shader stuff.

We will not be adding shadows or any other global illumination features. This is a CAD program that does not warrant photo-realistic rendering. There are much more important things to work on for the foreseeable future.

Evil-Spirit, 2022-09-30:
The lighting shader I made is a little bit strange. I can't recall all the details, but it is not regular diffuse. For good-looking black there is needed to introduce reflection component in a lighting model.

Evil-Spirit, 2022-09-30:
You can look at shader here https://github.com/solvespace/solvespace/blob/master/res/shaders/mesh.frag

phkahler, 2022-09-30:
@Evil-Spirit  I looked at the shader to make the ambient term configurable. It's just Snells law plus ambient. By increasing ambient and turning off directional lights we can do flat shading as requested by he who shall not be named.

I also have a branch where I added the fresnel term with some back lighting. It reduces the need to explicitly draw silhouette edges.

A good PBR shader would be great. I've got a repo started that has a decent shader.

Evil-Spirit, 2022-10-07:
> as requested by he who shall not be named.

Intoduce me please into this kind of things. Last time I made a mistake and another person waged a twitterwar against me (I was believing this person is "HE", but now is "SHE")