solvespace/solvespace #1222: How to quickly get started with the architecture of solvespace?

Nemo-HelloWorld, 2022-02-21:
### System information

- **SolveSpace version: 3.0~0e0b0252
- **Operating system: Windows 10 

I want to learn the architecture of solvespace, is there any relevant information that I can refer to besides reading the source code?

ruevs, 2022-02-21:
Few and far between are the "developer documents" :-)

https://github.com/solvespace/solvespace/tree/master/developer_docs
https://github.com/solvespace/solvespace/blob/master/CONTRIBUTING.md
https://solvespace.com/tech.pl
https://github.com/solvespace/solvespace/blob/master/exposed/DOC.txt
In many of the issues linked here https://github.com/solvespace/solvespace/issues/738 you will find deeper discussions.
https://github.com/solvespace/solvespace/issues/35
https://github.com/solvespace/solvespace/issues/389
https://github.com/solvespace/solvespace/issues/72
https://github.com/solvespace/solvespace/issues/386

closed 2022-02-21 by ruevs

Nemo-HelloWorld, 2022-02-21:
> Few and far between are the "developer documents" :-)
> 
> https://github.com/solvespace/solvespace/tree/master/developer_docs https://github.com/solvespace/solvespace/blob/master/CONTRIBUTING.md https://solvespace.com/tech.pl https://github.com/solvespace/solvespace/blob/master/exposed/DOC.txt In many of the issues linked here #738 you will find deeper discussions. In some of the issues linked here #1186 you will find deeper discussions. #35 #389 #72 #386

Thank you, very helpful :-)