solvespace/solvespace #1223: Crashes on Mac when I try to open preferences

sgall17a, 2022-03-03:
### System information

- **SolveSpace version:** Version 3.0 (3.0~54015b67)
- **Operating system:MacBook Pro (13-inch, 2020, Two Thunderbolt 3 ports) Monterey 12.1
### Expected behavior

Show preferences dialgo

### Actual behavior
program crashes

### Additional information
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SolveSpace [11508]
Path:                  /Applications/SolveSpace.app/Contents/MacOS/SolveSpace
Identifier:            com.solvespace
Version:               3.0 (3.0~54015b67)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-03-03 19:17:55.7666 +1000
OS Version:            macOS 12.1 (21C52)
Report Version:        12
Bridge OS Version:     6.1 (19P647)
Anonymous UUID:        851E8738-FAE2-D13A-4337-C0231C8B8A62

Sleep/Wake UUID:       E0E91738-BCEB-431A-A058-989DFCB7EEAF

Time Awake Since Boot: 89000 seconds
Time Since Wake:       626 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [11508]

phkahler, 2022-03-03:
That appears to be a development build from 3 years ago? Please try the [final 3.0 release](https://github.com/solvespace/solvespace/releases) or the latest [Development Edge build](https://github.com/solvespace/solvespace#via-automated-edge-builds)

ruevs, 2022-06-17:
@sgall17a have you tried the 3.1 release?

sgall17a, 2022-06-17:
I did try version 3.1 and the issue has gone.

Stuart Galloway
email: ***@***.***
Phone: 0421560031

closed 2022-06-17 by ruevs