solvespace/solvespace #1225: Cannot click on some constraints

jackfabulous, 2022-03-08:
### System information

- **3.0 0e0 b0252** 
- **Windows 10** 

### Expected behavior

I click on a constraint.  The constraint is then selected.

### Actual behavior

Frequently it is impossible to find a place to select a constraint because
1) the constraint is behind another constraint, so its "hit box" is completely obscured by another constraint.
2) the constraint is much smaller on screen than in v2x, and has a much smaller "hit surface"

### Additional information

phkahler, 2022-03-09:
We used to have an issue where the highlighted item (mouse over) might be different than what gets selected if you click. I assume this is not a recurrence of that problem. So you are unable to get the constraint in question to highlight (yellow) by moving the mouse over it, is that correct?

If you're working on a 2d sketch group, you might try using the new exploded-view either in the View menu, or just press "\". This shows all the sketch elements separated along an axis perpendicular to the sketch plane and was added to help in situations like this. You can set the "explode distance" at the bottom of the "view" panel in the text window. So press \ then rotate the view a bit to select your constraint and then \ again to put things back.

Another option might be to change the font size for constraints. I thought this was possible but I can't find any setting for it right now.

jackfabulous, 2022-03-09:
Unfortunately:
1) I cannot find an option to change the text size (which is much smaller than v2x)
2) pressing "\" seems to do nothing.
3) there is nothing in the view menu that is similar to "exploded view".  I tried random other menus from view, but am not turning anything up.  
4) I poked around in the reference section of the website, but don't see anything about an exploded view.

phkahler, 2022-03-09:
@jackfabulous you'll need to grab the latest "Edge" build for the exploded sketch option. Look here:
https://github.com/solvespace/solvespace#via-automated-edge-builds

ruevs, 2022-03-10:
> 1. I cannot find an option to change the text size (which is much smaller than v2x)

You can find all styles in the `line styles` section - at the bottom left of the text window. The default style for constraints is called `s006-#def-constraint`. Select it and change "text style | height" at the bottom.

ruevs, 2022-06-17:
@jackfabulous can you attach a file that demonstrates the problem?

wolfpackmars2, 2022-07-11:
I can attach an example.

Solvespace version 3.1~70bde63c
Linux Mint 19.3 (Ubuntu 18.04)

[image]
[nano_holder_bracket_2_r2.slvs.zip](https://github.com/solvespace/solvespace/files/9080121/nano_holder_bracket_2_r2.slvs.zip)

Within group g002-sketch-in-plane, there exists in the top left of the sketch several overlapped constraints that are un-selectable at any zoom level. Namely, the angle constraints between the line segments.

phkahler, 2022-07-11:
Press the backslash key @jackfabulous to explode the view. You will need to rotate the view so you're not looking straight at the sketch. Use the explode distance in the view menu to increase the separation if needed. This should allow you to select what you want.

ruevs, 2022-07-11:
@wolfpackmars2 selecting some constraints is indeed rather difficult sometimes. Your file is a nice example. Nevertheless it is possible to do it with some patience. Here is a video:

https://user-images.githubusercontent.com/15338069/178258886-2eacc864-cc4b-4069-9ccb-669e932bafe6.mp4

As @phkahler mentioned - using the `View | Show Exploded View` (`\`) option (available in version 3.0 and later) makes it even easier. Here it is done on your model:

https://user-images.githubusercontent.com/15338069/178259086-4e38beeb-83d0-4546-b87d-092443070619.mp4

closed 2022-07-11 by ruevs

ruevs, 2022-07-11:
By the way @wolfpackmars2 here is a "leaner" way to draw your model using rotates.
[nano_holder_bracket_2_r3_ruevs.zip](https://github.com/solvespace/solvespace/files/9084231/nano_holder_bracket_2_r3_ruevs.zip)

ruevs, 2022-07-11:
By the way @wolfpackmars2 here is a "leaner" way to draw your part using rotates:
[nano_holder_bracket_2_r3_ruevs.zip](https://github.com/solvespace/solvespace/files/9084286/nano_holder_bracket_2_r3_ruevs.zip)

wolfpackmars2, 2022-07-11:
Very cool! Knowing about back slash to expose the layers will come in handy as I become more proficient with solvespace.

[image]
(source: Hazure Skill)