solvespace/solvespace #1228: Build fails (on Windows) bc. of libpng when awk was found in path

Dromantor, 2022-03-31:
Just something I found when I tried to build solvespace some time ago and rediscovered when building it again today (in console with nmake).

Libpng has some (optional as it seems) awk build script that is executed if cmake finds the awk executable in path. For me this is the case because gawk is part of my cygwin installation which is in path.

This leads to the following build error:
[image]

If I remove awk from path (by renaming the cygwin directory temporarily) and delete the cmake cache the build succeeds without any problems.

I don't know if its just a windows+cygwin-gawk thing or it happens with other configurations too, didn't test extensively. I just though it might be good to mention it here because this could happen to other people too.

phkahler, 2022-03-31:
@Dromantor  Lots of questions. Why does the awk script fail for you? Can it be fixed upstream? Can we (should we) tweak ours somehow to make it not happen in this (relatively rare) scenario?

@ppd any ideas? Also, our snap builds seem to be not working in CI.

ppd, 2022-03-31:
This screams Windows line endings (\r\n) vs. Unix line endings (\n). So some generated header file has incompatible line endings, I guess. 

@phkahler I'll look into it.

ppd, 2022-03-31:
`pnglibconf.tf4` probably has CRLF, then gawk turns it into some unholy abomination as it only understands LF. It probably needs an extra conversion/filter step if run in this specific build environment (Windows + GNU awk). Or an awk version that supports CRLF.

Dromantor, 2022-03-31:
Installed Windows port from [http://gnuwin32.sourceforge.net/packages/gawk.htm](http://gnuwin32.sourceforge.net/packages/gawk.htm) and added it to path before cygwin. Error gone, works now. Line-endings really were the problem as it seems.
It would be good to mention this problem somewhere in the build instructions since gawk is part of the base install of cygwin.

phkahler, 2022-03-31:
@Dromantor  see if you think the updated MinGW instructions are adequate to close this.

Dromantor, 2022-03-31:
@phkahler I'm using the VS tools in console so its a general build-on-Windows thing and not just MinGW. Other than that the text is fine.

closed 2022-04-01 by phkahler