solvespace/solvespace #1229: Snap builds fail with "aa_is_enabled() failed unexpectedly"

ppd, 2022-03-31:
## What's the problem?

Snapd switched to strict AppArmor checking: https://github.com/snapcore/snapd/commit/c6011693a816f7f8a5b0c7858ddce91c6ef1a352

This check fails in containers, rendering our current pipeline useless because it relies on Docker to provide an arm64 environment for cross-builds.
The GitHub action's author has no idea how to fix this either: https://github.com/diddlesnaps/snapcraft-container/issues/8

## What can we do?
Either wait and pray, or drop the edge arm64 builds. We could still set up an arm64 outside GH actions, e.g. by syncing our repo to launchpad, which has arm64 workers or something. But in the short term, I propose dropping the arm64 edge build in CI.

@phkahler Does that sound reasonable to you?

phkahler, 2022-03-31:
> Does that sound reasonable to you?

Yes it does. You've been the one getting snaps together, so if you're OK with it so am I  ;-)  It would be nice to build for everything, but sometimes that's just not possible. Looks like there are people actively trying to solve the problem, but for now we should probably just drop it.

ppd, 2022-03-31:
I'll make a PR. Anything's possible with unlimited time, but at the moment I'm elbow deep in thesis work and don't have the mental energy to debug this complicated stack (GH actions + docker + snapd + QEMU user mode emulation).

PR #1232 (2022-04-13): CI: Use upstream action to build snap & drop arm64 snap
Snapd stopped working in Docker containers in commit
https://github.com/snapcore/snapd/commit/c6011693a816f7f8a5b0c7858ddce91c6ef1a352

Therefore, we cannot use the current build action,
which relies on Docker to implement cross-compilation.
Switching to the upstream build action avoids the problem at the cost
of losing the arm64 builds on amd64 hardware.
Those need to be reintroduced by some other mechanism,
e.g. via Launchpad, in the future.

Fixes #1229
commit 6a6fffed (2022-04-13): CI: Use upstream action to build snap & drop arm64 snap (#1232)

Snapd stopped working in Docker containers in commit
https://github.com/snapcore/snapd/commit/c6011693a816f7f8a5b0c7858ddce91c6ef1a352

Therefore, we cannot use the current build action,
which relies on Docker to implement cross-compilation.
Switching to the upstream build action avoids the problem at the cost
of losing the arm64 builds on amd64 hardware.
Those need to be reintroduced by some other mechanism,
e.g. via Launchpad, in the future.

Fixes #1229
closed 2022-04-13 by phkahler