solvespace/solvespace #1231: Constraint to line passing through work plane only decreases DOF count by 1

twhitehead, 2022-04-12:
### System information

- **SolveSpace version:** 3.0-v3.0
- **Operating system:** nixos 21.11

### Expected vs actual behavior

If you are sketching on a workplane and constrain on a line passing through that work plane, the DOF count is only decreased by 1 while it should be decreased by 2.

The following is a screen shot showing the simplest example I could think of if this: a point added to a work plane and constrained to a line passing through the work plane. It is entirely locked down, the DOF still shows 1.

[image]

twhitehead, 2022-04-12:
Right. I bet it is because it is constrained to the projection of the line in the plane, and, while technically a line that is perpendicular to the plane would reduce the DOF by 2, probably even perpendicular lines are perpendicular due to rounding....

ruevs, 2022-04-12:
Your "bet" is correct. From the reference:

> Most constraints are available in both 3d and projected versions. If a workplane is active, then the constraint applies on the projection of the geometry into that workplane. If no workplane is active, then the constraint applies to the actually geometry in free space. 

https://solvespace.com/ref.pl

closed 2022-04-12 by twhitehead