solvespace/solvespace #1233: Glitch with menus and application drawing on ARM

Thovthe, 2022-04-14:
### System information

- **SolveSpace version:** v3.0 built with PKGBUILD below 
- **Operating system:** ManjaroARM on the PinebookPro 
-**Architecture:** aarch64 (arm64/armv8)
<details>
  <summary>PKGBUILD (Expand)</summary>
```
# Maintainer: thovthe
# Contributor: lis
# Contributor: mis
# Contributor: oslik
pkgname=solvespace
pkgver=3.0
pkgrel=1
pkgdesc="SOLVESPACE is a lightweight, free parametric 3d CAD tool."
arch=('i686' 'x86_64' 'aarch64')
url='http://solvespace.com/'
license=('GPL3+')
provides=('solvespace'
	  'solvespace-cli')
conficts=('solvespace'
	  'solvespace-cli')
depends=('glibc' 'libspnav' 'libsigc++<3.0.0')
makedepends=('gcc' 'git' 'cmake' 'openmp' 'zlib' 'libpng' 'cairo' 'freetype2' 'json-c'
	     'fontconfig' 'gtkmm3>=3.16.0' 'libpangomm-1.4.so' 'mesa' 'glu')
source=("${pkgname}-${pkgver}::git+https://github.com/${pkgname}/${pkgname}.git#tag=v${pkgver}")
sha256sums=('SKIP')

prepare() {
    cd "${srcdir}/${pkgname}-${pkgver}"
    git submodule update --init extlib/{libdxfrw,mimalloc}
    mkdir build
}

build() {
    cd "${srcdir}/${pkgname}-${pkgver}/build"
    cmake .. \
          -DCMAKE_INSTALL_PREFIX=/usr \
          -DCMAKE_BUILD_TYPE=Release \
          -DENABLE_OPENMP=ON \
#	  -DCMAKE_INSTALL_LIBDIR=/usr/lib/${pkgname} \	  
#	  -DENABLE_LTO=ON
    make
}

package() {
    cd "${srcdir}/${pkgname}-${pkgver}/build"
    make DESTDIR="${pkgdir}" install
}

```
</details>

### Expected behavior
Application stops drawing menus when you click into the workarea and doesn't exhibit other graphical glitches.

### Actual behavior
Ghost images of the menus getting chopped up by the wires representing planes while view is rotated or panned.
When switching to SOLVESPACE from other applications the other application's window content is drawn in one of the windows.

See attached pictures.

### Additional Information
This may not be solvespace's fault but none of my other applications behave like this.  I'll continue to fiddle.

ruevs, 2022-04-14:
What is reported on the bottom of the configuration screen (OpenGL info)?

Thovthe, 2022-04-14:
[image]

ruevs, 2022-04-14:
Is there any other windowed OpenGL software that you can try? This is just a wild guess... I suspect you are the first one to try SolveSpace on this platform.

Thovthe, 2022-04-14:
> Is there any other windowed OpenGL software that you can try?
Quite possibly. I'm not sure how to query pacman for build dependencies though.  Maybe you know of some programs that I might have access to on linux?

> I suspect you are the first one to try SolveSpace on this platform.
Very probably, it took me a couple of hours to get a working PKGBUILD

Thovthe, 2022-04-14:
glxgears runs fine but it has no menus and only one window.

ruevs, 2022-04-15:
The only other ARM Linux platform (that I know of) that SolveSpace has been tried on is Raspberry Pi - it works fine but there is a glitch in the menu bar (not the menus themselves). See here https://github.com/solvespace/solvespace/pull/825#issuecomment-734899712 

Would the ARM64 build from the Snap store work? https://snapcraft.io/solvespace

Another thing to try may be to build it with the old OpenGL 1 renderer and see how that works (set OPENGL=1 in CMake). But I am not even sure if it still works on Linux. (https://github.com/solvespace/solvespace/pull/424)

That said I'm a simple user under Linux and it's not my daily platform - so I don't have any other ideas.

phkahler, 2022-04-15:
I wonder if @rpavlik knows anyone working on the panfrost driver. This could be a driver issue?

Thovthe, 2022-04-15:
@ruevs
It sounds like that might be a related glitch.

> (set OPENGL=1 in CMake)

Can I pass this as a command line option somehow?  (not a C programmer)

I'm somewhat allergic to snap for ideological reasons but if the rebuild is not illuminating I could try that.

Thovthe, 2022-04-15:
Update on how to reproduce:
I was having trouble reproducing the menu behavior.  Seems that the application rendering gets buggy if you somehow get the viewport to change while menus are open.  `Select a menu` then hit `w` while off axis

rpavlik, 2022-04-17:
I know people on my team have more interesting sbc boards than raspberry pi, so they can try to repro and pass info along to the panfrost team

Thovthe, 2022-04-18:
Let's say cutting edge, Manjaro lags upstream for stability's sake.

phkahler, 2022-04-19:
> glxgears runs fine but it has no menus and only one window.

And is probably one of the first programs the driver developers tested with ;-)

BTW assuming you're running on X and not Wayland. Either way, the blocky garbage on top of your web browser strongly suggests something lower level to me.

Thovthe, 2022-04-19:
> > glxgears runs fine but it has no menus and only one window.
> 
> And is probably one of the first programs the driver developers tested with ;-)
> 
> BTW assuming you're running on X and not Wayland. Either way, the blocky garbage on top of your web browser strongly suggests something lower level to me.

So that looks at first glance like it's on top of my webbrowser but that's a solvespace window (i had alt-tabbed to solvespace), you can tell because the window decoration is not for firefox.  Or do I misunderstand what you are saying?

I am running XFCE on x11.

It's only solvespace that retains the (buffer?) screen content from other applications.  No other applications I've tried have this drawing issue either from solvespace or other applications.

ruevs, 2022-07-05:
@Thovthe were you able to build it with OpenGL1?
The cmake command should be: `cmake .. -DCMAKE_BUILD_TYPE=Release -DOPENGL=1`