solvespace/solvespace #1239: Multiple "Message" dialogs when deleting elements with dependant constraints (minor, UI)

gnbl, 2022-05-10:
Deleting elements that other elements are constrained to, this message is shown:

```
---------------------------
SolveSpace - Message
---------------------------
Additional sketch elements were deleted, because they depend on the element that was just deleted explicitly.

These include: 
     0 requests
     2 constraints
     0 groups

Choose Edit -> Undo to undelete all elements.
---------------------------
OK   
---------------------------
```

Unfortunately I can't provide a minimal example of this, but sometimes this automatic deletion appears to cause another automatic deletion and a corresponding message dialog. I've only noticed because the second message takes the focus, so hitting Return to accept with `OK` does not work and I have to use the mouse...

This could be resolved by restoring focus to the other dialogs, or by catching all automatic deletions/dependencies before showing the dialog.

phkahler, 2022-05-10:
@gnbl what OS is this on? I thought those message boxes were modal so there couldn't be 2 at the same time. An example could be helpful.

gnbl, 2022-05-10:
windows-openmp 3.0~b334996c on Windows 10 64 bit.

Sorry, I tried to come up with an example, but unsuccessful so far.
Next time it happens, I'll take a closer look and will attempt to reproduce in minimal example file.

gnbl, 2022-05-11:
Here's what it looks like, even three dialogs:
[image]
Cannot reproduce in example sketch.

iscgar, 2025-05-16:
So, the cause for this is quite simple once it is reproduced:

[image]

Calling `Message()` causes events to be handled, and because of that, the timer may fire with a scheduled regeneration after entity splitting, causing a call to `GenerateAll()` while we're showing the message box, but before it's dismissed. Because we still didn't zero the `deleted` member, the code path happily shows the message box again, which can lead to another scheduled regeneration to be fired, etc.

https://github.com/solvespace/solvespace/blob/2c5bb5d475e03a73da685649d2009539bfa96a7e/src/generate.cpp#L331-L352

I'll open a PR shortly to fix this issue.

closed 2025-05-17 by phkahler

ruevs, 2025-05-17:
Wow! This was easy (in retrospect) :-)

PR #1590 (2025-05-17): generate: avoid message box avalanche on Windows during regeneration
Calling `Message()` can cause processing of a scheduled dirty regeneration while the message box is still shown. Since the deleted stats are only zeroed after the message box is dismissed, this can cause the message box to be shown multiple times.

Fix it by changing the code to reset the deleted stats and use a saved copy for showing the message box.

Fixes #1239.
commit 322dbece (2025-05-17): generate: avoid message box avalanche on Windows during regeneration

Calling `Message()` can cause processing of a scheduled dirty regeneration
while the message box is still shown. Since the deleted stats are only
zeroed after the message box is dismissed, this can cause the message
box to be shown multiple times.

Fix it by changing the code to reset the deleted stats and use a saved
copy for showing the message box.

Fixes #1239.