solvespace/solvespace #1240: Reference plane visibility (feature)

gnbl, 2022-05-11:
Windows 10 openmp 3.0~b334996c 

The reference planes are centered on the origin and displayed proportionally to the view (window) size, regardless of zoom.
This is helpful for orientation when orbiting in 3D view since the full planes are always visible (around the origin).

When zooming in on elements at a distance from the origin, the reference planes are not visible anymore.

Two ideas:

- show infinite reference planes (only works in 2D / view aligned to planes)
- show reference planes within the bounding box of elements (also works in 3D)

ruevs, 2022-06-14:
How do you envision the reference planes being drawn away from the origin or from their intersection lines? What would be visible?

A related topic is this https://github.com/solvespace/solvespace/issues/1006

gnbl, 2022-06-18:
Here's an example of what I imagined for my second idea (imagine dotted lines):
[image]

It would extend the reference planes beyond the bounding box. But this mixes screen and geometry.
Maybe it's simpler to resize the reference planes so their intersections are always visible on screen? However, that might be confusing a little since they would not be square and change appearance dynamically.