solvespace/solvespace #1241: Right mouse button (RMB) cannot be used for panning when editing a dimension (enhancement)

gnbl, 2022-05-11:
windows 10 openmp version 3.0~b334996c.

When (automatically) editing a (new) dimension that happens to be off screen (e.g. angle constraint), panning with RMB (right mouse button) does not work, since the dimension input field has focus.

Two ideas:

1. RMB cancels editing, but then only the second attempt to pan will work
2. RMB pans, while edit field remains focused

phkahler, 2022-05-11:
I'm not sure this is really feasible, and it would have to work on all platforms. Just press enter, pan, and then double click the dimension.

closed 2022-05-11 by phkahler

gnbl, 2022-05-11:
But you could check whether it is feasable, just to be sure, before shutting this suggestion down?

phkahler, 2022-05-11:
> But you could check whether it is feasible, just to be sure, before shutting this suggestion down?

No, I could not.

It takes a lot of time to investigate UI toolkits that I'm not familiar with, and for the very minor benefit it's just not worth it. I have a large backlog of things that I actually *want* to do.  If someone submits a PR for something like this I'm happy to review it and even merge it, but not me nor any other semi-regular contributors are going to jump on this and take it on.

gnbl, 2022-05-11:
That's perfectly fine, and you could just say so.